silvanmelchior/RPi_Cam_Web_Interface #319: Error : [EOF when reading a line]

gpapig, 2017-01-06:
Hi !

I've an issue when motion detect a movement, the detection is OK, it starts to record the file (as per the button status, but once finished I got the below message : Error: [EOF when reading a line] (when a terminal is open on my pi). The video is finally saved (take a few times, probably due to my Pi0)

Any idea?
Everything's is up to date, lateste raspbian lite.

find below the lastest lines in scheduleLog.txt

> {2017/01/06 14:44:54} Finished boxing /var/www/cam/media/vi_0001_20170106_144439.mp4 from Box Queue at pos 1
> {2017/01/06 14:44:54} Removed item from Box Queue
> {2017/01/06 14:44:55} External motion detection stopped
> {2017/01/06 14:44:55} recommended video buffers 1
> {2017/01/06 14:44:55} h264 buffers set to recommended 1
> [2017/01/06 14:44:56] Stop capture request ignored, already stopped
> {2017/01/06 14:46:18} External motion detection started
> {2017/01/06 14:46:19} recommended video buffers 1
> {2017/01/06 14:46:19} h264 buffers set to recommended 1
> {2017/01/06 14:46:31} Capturing image
> [2017/01/06 14:50:56] Stop capture request ignored, already stopped
> [2017/01/06 14:51:48] Start capture requested from Pipe
> [2017/01/06 14:51:48] Send ca 1
> {2017/01/06 14:51:48} Capturing started
> [2017/01/06 14:52:05] Stop capture requested
> [2017/01/06 14:52:05] Send ca 0
> {2017/01/06 14:52:05} Capturing stopped
> {2017/01/06 14:52:05} Add /var/www/cam/media/vi_0002_20170106_145148.mp4 to Box Queue at pos 2
> {2017/01/06 14:52:05} Executing macro /var/www/cam/macros/end_vid.sh "/var/www/cam/media/vi_0002_20170106_145148.mp4.h264"
> {2017/01/06 14:52:11} Start boxing /var/www/cam/media/vi_0002_20170106_145148.mp4.h264 to /var/www/cam/media/vi_0002_20170106_145148.mp4 Queue pos 2
> {2017/01/06 14:52:13} Finished boxing /var/www/cam/media/vi_0002_20170106_145148.mp4 from Box Queue at pos 2
> {2017/01/06 14:52:13} Removed item from Box Queue

roberttidey, 2017-01-06:
The error messages are coming from the external motion process and I think they are occuring during start up of the motion process which happens turning motion detection on occurs and it reads its config file (/etc/motion/motion.conf. It sounds like this config file may be out of sync with the motion version maybe because the motion version has been updated. You could check the motion.conf file for both a missing line terminator at the end of the file and also for that config value sdl_threadnr. You could also attach the motion.conf file here so I could look at it.

You could also switch to using the internal motion detect process (under camera settings). I use this myself exclusively. It uses vector data from the camera and avoids running the motion process altogether so these errors associated with motion wouldn't occur.

The second capture in the log looks fairly normal. The 7 second video is converted to mp4 quite quickly. There is, however, a delay of a few seconds in starting the conversion (box) process which I don't understand at the moment. In my logs from Pi Zero cameras the box operation starts within a second of end of capture. Have you got other stuff running on the Pi?

roberttidey, 2017-01-07:
The sdl_threadnr config parameter seems to be a new upcoming feature for the external motion process. See http://www.lavrsen.dk/foswiki/bin/view/Motion/ConfigOptionSdlThreadnr

For some reason the base conf file that motion is now creating includes the parameter even though that version of motion does not support it. That is what triggers the error message. It is quite harmless and can be ignored. If you want to get rid of it then find the line with sdl_threadnr in the /etc/motion/motion.conf file and delete it.

As I say I use internal motion detection and so do not see this error message anyway.