silvanmelchior/RPi_Cam_Web_Interface #320: Audio support?

czarina, 2017-01-07:
Hello, I would really like to use RPi-Cam-Web-Interface, but I need to stream audio as well as video. 

Is this possible with some hacking on RPi-Cam-Web-Interface, or is there any other package you'd recommend?

roberttidey, 2017-01-07:
There are two different aspects to audio support.

The first is live streaming of audio + video which you seem to be primarily interested in. The second is incorporating audio into any video recordings that are made.

konstantinj on issue #250 gave macros which achieved the second aspect using macros. Here the start video macro kicks of an audio recording to a file and the end video macro arranges for that to be added into the mp4 recording during the boxing process. This gives a combined video + audio recording which would naturally playback. The sync in the recording will be dependent on whether video buffering is turned on. This is turned on to give pre-motion trigger recording and will make it difficult to get good sync in the recording. If the buffer is 0 then the sync should be pretty good. I haven't seen any c code changes yet but obviously this would make it easier to turn this support on.

The first aspect of live audio + video is different. The video in the current stream is a mjpeg stream fed to a video window. It should be possible to have a separate audio stream fed to the browser in parallel. The sync should be fairly good with this. It would require the raspberry server end to have an audio stream server linked to the microphone and the main web page modified to link to the audio stream at the server.

You might also want to check out uv4l based projects as these are more targetted at pure steraming solutions incorporating audio and video.

alfem, 2018-09-26:
Maybe with an Icecast? https://stmllr.net/blog/live-mp3-streaming-from-audio-in-with-darkice-and-icecast2-on-raspberry-pi/

I tested it months ago, and It can stream audio from the usb microphone (using Darkice). I am still looking for a simple method to play this audio stream in a web page.

viveksanagari, 2019-01-21:
Hi,
If a video and audio solution for raspberryPi is necessary, an out-of-the-box solution is available at https://github.com/iizukanao/picam-streamer

The SD card image is attached in the above link for the solution. However, all the camera settings and options which are present in RPI_Cam_web_interface and not available in this. This solution provides a ready to use video and audio live streaming without any changes required.

elgladiadordelcielo, 2024-01-15:
Con Icecast o Darkice la latencia en el sonido es excesiva. Yo lo estoy solucianando con UV4L y usando una versión antigua (Stretch) del sistema operativo compatible son el repositorio que hay que añadorle para descarga. A mí me ha costado la misma vida encontrar la información y es por eso que creo necesario compartirla.