silvanmelchior/RPi_Cam_Web_Interface #321: File delete by macro cannot be deleted by macro

florisjan78, 2017-01-08:
Hello,

Not 100% sure if this is a RPI_cam_Web_Interface problem or just something I'm doing wrong in my scripts.

I use the 2 following macro scripts as explained in the instructions, to create and delete a file at motion start / end. It is a network mount to another pie's /dev/shm.

File: /var/www/html/macros/start_vid.sh
`#!/bin/bash
echo "MOTION" > /mnt/piramdrive/activity.fd
`
File: /var/www/html/macros/end_vid.sh
` #!/bin/bash

while [ -f /mnt/piramdrive/activity.fd ];
do
        echo "deleting activity.fd" >> /mnt/usbdrive/log/bash.log
        rm -f /mnt/piramdrive/activity.fd >> /mnt/usbdrive/log/bash.log 2>$
        sleep 1
done
`
resulting in:
deleting activity.fd
rm: cannot remove ‘/mnt/piramdrive/activity.fd’: Operation not permitted

This is the directory listing where the file is created:

pi@pie2:~ $ ls -l /dev/shm/
total 92
-rw-r--r-- 1 pi       pi          20 Jan  8 22:27 light.fd
drwxrwxrwx 2 www-data www-data   100 Jan  8 22:42 mjpeg

The file appears to be created by the user pi, it works fine as soon as motion is detected. The deletion however goes wrong, is the script run by the same user?

I do the same kind of thing from python scripts, but they are started from rc.local, so user root.

This is the macro directory listing:

pi@pie1:~ $ ls /var/www/html/macros/ -l
total 16
-rwxrwxrwx 1 www-data www-data 210 Jan  8 22:25 end_vid.sh
-rwxrwxrwx 1 www-data www-data  33 Dec 24 00:39 error_hard.sh
-rwxrwxrwx 1 www-data www-data  61 Jan  7 19:14 start_vid.sh
-rwxrwxrwx 1 www-data www-data  42 Dec 24 00:39 test.sh

Thanks in advance,

Fj

roberttidey, 2017-01-08:
This sounds like a permissions issue. The macros run as user www-data (which is the user the web server uses and allows better control of security given that these macros are triggered by web operations.). This means that any file operations should be on files or directories that user www-data has relevant permissions.

I'd suggest first testing by doing the operations on a local folder which you have explicitly given www-data access to. If that shows the problem is permissions then there are a number of options; the easiest being to give www-data access to your remote folder.

florisjan78, 2017-01-09:
It is indeed. Creating / deleting it in /dev/shm/mjpeg solves the problem. Not sure how to fix it over a network mount though, but this will work fine. Thanks.

closed 2017-02-16 by florisjan78