silvanmelchior/RPi_Cam_Web_Interface #322: More than 9999 images in a timelapse sequence

pishotta, 2017-01-16:
Hi,

Is there a way to have raspimjpeg sequence its pics using 5 decimal places instead of 4 (packed with leading zeros).  Perhaps fiddling with lapse_path to achieve names like:

tl_0001_00000_20170116_143358.jpg, through 
tl_0001_99999_20170116_143358.jpg

This sometimes causes sorting problems when N exceeds the default of 4 digits.

Thanks,

-- Fred P.

roberttidey, 2017-01-17:
The current software has 4 digits as the starting format and this expands as required. This was not changeable.

I have now added support for a config variable in /etc/raspimjpeg to control this. The default is %04d which gives the current scheme. If this is changed to %05d then you get 5 digits as the start point.

This in the latest  version uploaded.

pishotta, 2017-01-17:
Perfect!  Thank you.

-- Fred P.

pishotta, 2017-01-19:
Tested it with N > 11000 stills that I ran through gst to animate.  Worked well with no unintended consequences detected so far.  Was a little worried about file name parsing going on in preview.php and config.php, where there appears to be some hard-coding of length characteristics and specific expressions of "%05d" in some places, but no problems in the processing I performed.  Will worry again if N approaches 99999, but that's unlikely in my remaining years.  Thanks for the quick response. The few scripts I wrote to workaround the issue are now unneeded.  I can return to the pure feed.  -- Fred P.

closed 2017-01-21 by pishotta