silvanmelchior/RPi_Cam_Web_Interface #323: Stream halted - unexpected failure

florisjan78, 2017-01-20:
All the sudden after running for 4 days and 12 hours, the camera dimensions were altered and the stream was halted. The last frame from /dev/shm/ is black, does contain the annotation and truncated at 288 lines.  I'll provide you as much information as possible.

RPi Cam Control v6.2.21

The camera settings seem to have gone to default as my custom settings have disappeared from the uconfig file, the following lines are gone:
video_width 1296
video_height 972

the last part from the scheduler log:

{2017/01/20 10:58:21} Capturing image
{2017/01/20 11:00:16} Change: video_width = 1920
{2017/01/20 11:00:16} Change: video_height = 1080
{2017/01/20 11:00:16} Change: video_fps = 01
{2017/01/20 11:00:16} Change: MP4Box_fps = 30
{2017/01/20 11:00:16} Change: image_width = 2592
{2017/01/20 11:00:16} Change: image_height = 1944
{2017/01/20 11:00:16} recommended video buffers 1
{2017/01/20 11:00:16} h264 buffers set to recommended 1
{2017/01/20 11:00:16} Set up internal detect width=121 height=68
{2017/01/20 11:00:17} DEBUG 3
{2017/01/20 11:00:34} Capturing image
{2017/01/20 11:00:49} Timelapse stopped
{2017/01/20 11:00:51} Internal motion detection stopped
{2017/01/20 11:00:51} recommended video buffers 1
{2017/01/20 11:00:51} h264 buffers set to recommended 1
{2017/01/20 11:00:51} Set up internal detect width=121 height=68
{2017/01/20 11:00:52} DEBUG 3
{2017/01/20 11:00:53} Stream halted
[2017/01/20 11:24:11] Scheduled management tasks. Next at 1484911451

Raspimjpeg is running: (top output:)

546 www-data  20   0   79452   2276   1484 S  0.3  0.9 635:59.91 raspimjpeg

The motion macro was not running.
My own python script was, which copies the cam.jpg file to another pi's ramdrive. There were no mount issues or anything else that could cause problems. These problems never cause rpi cam to fail anyway, because it leaves the system alone. Some settings are changed when lighting levels changed using the FIFO11 file, the last time this happened was a couple of hours earlier.

From the scheduler log:

{2017/01/20 09:02:47} Timelapse started
{2017/01/20 09:02:48} Change: brightness = 50
{2017/01/20 09:02:49} Change: exposure_mode = auto
{2017/01/20 09:02:50} Change: white_balance = auto

And the part of my python script that does the file copying:

if (datetime.now()-lastFrame).total_seconds() >.5 and not os.path.isfile('/mnt/uberpiramdrive/frame-monitor.jpg'):
	lastFrame=datetime.now()	
	copyfile('/dev/shm/mjpeg/cam.jpg', '/mnt/otherpiramdrive/frame-monitor.jpg')

Regards,

Fj

roberttidey, 2017-01-20:
The logs would indicate that the FIFO pipe received and acted on a series of commands that set the video parameters (px command), stopped time lapse operation (tl 0), stopped motion detection (md 0), and finally halted the camera (ru 0).

There is nothing in raspimjpeg that could do that by itself. 

I'm not sure what could trigger this. One could certainly simulate this log by pressing a sequence of commands in the client but I guess that wasn't the cause. There are no scheduler log events round this time so they are not coming from there. Either some external process went wrong and fed in some random commands, a web client javascript crashed in some peculiar way and self issued a set of commands, or the web server managed to send out a spurious stream of commands. 

It would be interesting to know if it could be recovered by doing a web Camera start button push as that would tend to indicate the base system was OK but just responded to a set of commands.

Also was a web client running when this happened?

One might have to see if one can get logging from the web client or the lighttpd server. I don't use the that server myself so don't know what logging is available there.

florisjan78, 2017-01-20:
The only instructions issued by me are the ones shown in the log, there are no other scripts or programs that send commands. There were no clients active as far as I know, the access log is disabled. Turning it on would probably kill the SD card. I'm not sure if the camera start button was available, but should this ever happen again I will try it.

Quoting the commands again....

{2017/01/20 11:00:16} Change: video_width = 1920
{2017/01/20 11:00:16} Change: video_height = 1080
{2017/01/20 11:00:16} Change: video_fps = 01
{2017/01/20 11:00:16} Change: MP4Box_fps = 30
{2017/01/20 11:00:16} Change: image_width = 2592
{2017/01/20 11:00:16} Change: image_height = 1944

...would seem to me like quite a few and quite specific instructions to be issued. It almost seems as if somebody was messing around with the settings and then left it. Like the 1fps and 30 mp4, they seem odd to me.

Fj

roberttidey, 2017-01-20:
Changing width height etc is actually just one command (px) plus the parameters. The web client sends this command when you change any of the numbers.

The parameters with the 1fps are actually those used by the client camera setting preset called Full HD timelapse. Tis records video at 1fps then boxes to 30fps to give a X30 speed up. So it is like that preset got triggered somehow.

You might be able to position the log into /dev/shm or /tmp which are in RAM rather than on the SD card.

florisjan78, 2017-01-21:
Yes I thought about that but unfortunately the lighttpd log logs every download resulting in enormous logfiles when the stream is open in a client. The pi is in an outdoors camera so I can't just attach an external hard drive for the log. 

I would like to add then that at 1 video fps the camera can not adjust to the lighting and keeps going from full brightness (showing as green) to dark once a second, sometimes flashing a bit. I think I have read about this before somewhere on this forum.

Thanks for looking into this, if I learn more I will let you know.

Fj

pishotta, 2017-01-21:
A reminder:  marginal power input to the RPi can trigger some truly funky behavior.  Don't know about your setup, but keep that in mind.

gstrauss, 2017-02-01:
> The last frame from /dev/shm/ is black, does contain the annotation and is truncated at 288 lines.

Is /dev/shm filling up?  Is /tmp filling up?

> Yes I thought about that but unfortunately the lighttpd log logs every download resulting in enormous logfiles when the stream is open in a client. The pi is in an outdoors camera so I can't just attach an external hard drive for the log.

FYI: You can configure lighttpd to log to a pipe, and write a small program which listens to the pipe and sends the log elsewhere on the network, e.g. cloud or rsyslog or ...

florisjan78, 2017-02-16:
It has not happened ever since so'll I'll close the issue.

closed 2017-02-16 by florisjan78