silvanmelchior/RPi_Cam_Web_Interface #325: Add ru command to the do_cmd.sh macro (switch IR leds on/off)

WillemJann, 2017-02-10:
Hello,

My camera setup includes some IR leds which I want to enable when the camera is enabled (during the night), and disable when the camera is disabled. My schedule has a AutoCamera_Interval of 10, so there is a regular on/off switching of the camera using the `ru` command.

I think the most solid way to make sure the IR leds are always in sync with the camera, is using the macro do_cmd.sh. Therefore I would like to perform some actions (switching the leds on/off) on a camera shutdown (`ru 0`) and startup (`ru 1`). But according to the documentation, the do_cmd.sh macro is currently only fired on the `md` command.

Can you make the `ru` command also fire the do_cmd.sh macro? And can you provide an example script how I can detect the `ru 1` and `ru 0` actions using the parameter of this script?

Thanks in advance!

roberttidey, 2017-02-10:
The updated version should be operational now for ru commands but I haven't tested it yet.

The do_cmd.sh macro is called with the contents of the command buffer as a parameter so $ parameters can be used within the macro to determine the command and the parameters used which invoked the macro.

WillemJann, 2017-02-11:
Just tested the new version, and I can detect the `ru` command. Thank you!

closed 2017-02-11 by WillemJann