silvanmelchior/RPi_Cam_Web_Interface #327: Query current raspimjpeg activities and settings in effect

pishotta, 2017-02-11:
Hi,

Is there some command that queries the running raspimjpeg to see what it is currently doing? Like performing a capture (or not), or engaging in a timelapse collection (or not), or what the exposure settings currently are, or etc...? Maybe by specifying a command or variable name and returning its current operational value, or maybe just performing a dump of it all to STDOUT or some file?  In many cases these can be indirectly deduced from raspimjpeg.conf and uconfig, but not always reliably,  particularly not those collection activities currently in (or not in) progress -- or so it seems.  Shuffling through the documentation, I can't find any specific command to do this.  Looking at the behavior of the web pages, however, it appears they do have access to such info, but is not clear to me how it is obtained or if there is a consistent general approach to take.  But before I get too deep into the PHP files, thought I should ask, as it would be useful for external scripting/monitoring and/or the debugging thereof.

Thanks,

-- Fred P.

roberttidey, 2017-02-11:
The state of the program is maintained in the status file status_mjpeg.txt. This physically resides in /dev/shm/mjpeg but a link to it is made in the web folder. It can be read by local applications or via the web using the status_mjpeg.php. This is what the client does.

The file just has 1 text line with a keywords indicating what the application is currently doing
Error
halted
image
tl_video
video
tl_md_video
md_video
timelapse
ready
tl_md_ready
md_ready

pishotta, 2017-02-12:
Thanks.  I thought something had to be somewhere.  I just missed finding it.  I will note that during a timelapse sequence it oscillates between "timelapse" while it's waiting and "image" when it's capturing the image, but for my purposes I can work with that.

-- Fred P.

closed 2017-02-13 by pishotta