silvanmelchior/RPi_Cam_Web_Interface #329: PRi_Cam_Web_Interface not cloning

caesarx86, 2017-02-23:
Hi,

I'm having an issue installing the PRi_Cam_Web_Interface.
When i use the command on my Raspbian Jessie its not really doing anything:

[image]

Any idea what my problem can be?

Thanks in advance!

Nick

roberttidey, 2017-02-23:
Looks like the raspberry  is not reaching the internet for some reason.

Try ping github.com

To see if that can see the github site. CTRL-C will stop the pinging

caesarx86, 2017-02-23:
Internet works fine and github is accessible:

[image]

roberttidey, 2017-02-23:
I haven't seen a problem with the cloning part before. This is what the normal ouptut and I just rechecked it.

git clone https://github.com/silvanmelchior/RPi_Cam_Web_Interface.git
Cloning into 'RPi_Cam_Web_Interface'...
remote: Counting objects: 2330, done.
remote: Compressing objects: 100% (11/11), done.
remote: Total 2330 (delta 2), reused 0 (delta 0), pack-reused 2316
Receiving objects: 100% (2330/2330), 3.88 MiB | 292.00 KiB/s, done.
Resolving deltas: 100% (1382/1382), done.
Checking connectivity... done.

Does it create a PRi_Cam_Web_Interface folder?

I'll check to see if it is possible to get any logging on the git clone process

roberttidey, 2017-02-23:
Couple of other thoughts. I suspect this is not specific to this github repository but with any git clone process. It might be worth checking trying a different clone like wiringPi to see if that is the case.

Are you using a wifi connection or direct lan? There have been some reports of the routing getting confused if both are active and stopping the clone from proceeding. If you can try with just a wired connection.

Other users traced this down to a router problem which was resolved by rebooting their router.

caesarx86, 2017-02-24:
I did a complete re-install of Raspbian and now it works.
Thanks!

closed 2017-02-24 by caesarx86