silvanmelchior/RPi_Cam_Web_Interface #330: Timelapse convert feature does not work

brunoaduarte, 2017-02-23:
I've started the timelapse function in the web panel and recorded a few images. Then i tried to click the convert button, it showed the "Busy" status and then the converted video link. But when i list the files at "/var/www/media/" it shows a zeroed file.

I'm using latest

> RASPBIAN JESSIE LITE
> Minimal image based on Debian Jessie
> Version:January 2017
> Release date:2017-01-11
> Kernel version:4.4

I've manually installed "gstreamer" as it't not installed by default

> sudo apt-get install gstreamer1.0

Anything else i should do  ?

pishotta, 2017-02-23:
You don't need the whole gstreamer1.0 package (it's huge).  Only needs gstreamer1.0-tools (smaller) which is installed with the install.sh script.  Probably doesn't hurt, though.

gstreamer-whatever  is temperamental.  Start with smaller image resolutions.  It'll abort if they're too big.  I use 1024x768.

-- Fred P.

roberttidey, 2017-02-23:
Jessie full includes the normal gstreamer package but not the tools which convert needs and that is why the installer puts that piece in.

Jessie Lite does NOT include the gstreamer package. The tools piece is not sufficient. The easiest way with Lite is to install gstreamer 1.0 but this does also install quite a lot of other stuff which is not needed. It is possible to do a more selective install of just the particular components but that is quite a bit more complicated. So my advice is either to use full Jessie or put the gstreamer1.0 onto Lite. I normally do a full Jessie and then remove a lot of unrequired stuff which gets it back close to a Jessie Lite footprint. This is covered in the wiki, A single command can get rid of a lot of stuff.

As to the conversion process itself, the gstreamer method uses the GPU processor to do the conversion. A CPU only software conversion is very slow on the Raspberry Pi. Using the GPU does introduce some limitations. In particular high resolution formats above 1920 x 1080 will not work. Normally this is not too much of a restriction for time-lapse work as one wants videos that are normal video format (4 x 3, 16 x 9). If one wants to use very high resolution images then it is best to download the image set and convert using available tools on a PC or Mac.

brunoaduarte, 2017-03-02:
Ok, thanks @pishotta and @roberttidey

closed 2017-03-02 by brunoaduarte