silvanmelchior/RPi_Cam_Web_Interface #332: Can RPi_Cam_Web_Interface be made to work in read only mode?

igauravsehrawat, 2017-03-01:
Hi,
Thank you for the excellent software.
I want to make my raspberry pi to operate in read only mode. I followed this tutorial https://hallard.me/raspberry-pi-read-only/ . It works but `RPi_Cam_Web_Interface` ceased to work due to read only permissions.

1. Can you give some pointers if software can be modified to work in read only mode?
2. Is it possibility if one is using pi camera?

Should I make a separate partition to make it work?
Thanks

roberttidey, 2017-03-02:
There are a number of issues that would need to be dealt with to use this when installed on a read-only setup.

1) Media folder would need to be moved onto writeable medium if recordings were to be made. This can be achieved by linking to USB stick, drive or network.

2) 3 files are getting updated fairly continuously (preview jpg, status file and log). These are already in tmp but linked to from web folder. One would need to check the links still allowed writing.

3) Changes to camera settings are saved in a uconfig file in the web folder. One would either need to freeze the config when in read only mode or link this file out to a writeable location.

4) The control pipe FIFO is located in the web folder. This allows commands to be written to the system. It could be configured via /etc/raspimjpeg to be somewhere else.

One strategy would indeed to set up a separate medium and install the whole camera software there.

igauravsehrawat, 2017-03-02:
@roberttidey Thanks a lot for the reply.

I shrinked existed OS and made a separate partition on SD card.
That will do, right? I will take your pointers and try to implement them. 

Thanks

igauravsehrawat, 2017-03-02:
I followed this[0] blog, camera is working fine and saving upto time images. But streaming isn't working. Seems like problem with nginx.

[0]: https://hallard.me/raspberry-pi-read-only/

roberttidey, 2017-03-02:
From a browser you should be able to do something like http://ipaddress/folder/cam_pic.php  (where folder is any subfolder you installed in like html or blank if you installed at webroot.

This should return the latest static image. If that works then make sure you have default streaming selected first. Button under system should say MJPEG-Stream. That means the streaming should be working by fetching individual images.

The other thing to try is to turn on development tools option in your browser e.g. in Chrome. This will then show activity and any errors that are occurring.

igauravsehrawat, 2017-03-04:
Thank you for replying.

I figured out the issue. So it was with `nginx` service, it was unable to start since it was unable to write logs on read only disk, which i found using `nginx -t` command. So I had to sym link the nginx logs folder to `tmpfs` folder. I do this in the `nginx.service` file using `ExecStartPre` directive. On top of that I have separate partition which I mount on bootup for writing config files and data.

That's it. Problem solved.

Thanks

closed 2017-03-04 by igauravsehrawat