silvanmelchior/RPi_Cam_Web_Interface #333: Your installation script rewrote my nginx configs

pikami, 2017-03-08:
Your installation script rewrote my nginx configs.

pikami, 2017-03-08:
Oh shit.
That's not all, the whole www dir is now empty.
Well good job! Lost a shit load of work because of you, fix you fucking shit.

GaryLittlemore, 2017-03-08:
How rude, I'd tell you to sling your hook if you spoke to me like that.

pikami, 2017-03-08:
@GaryLittlemore I'm just really frustrated because I lost a lot of work.
I know that my rude speech wont fix anything, but I hope that other people can learn from my mistakes

pishotta, 2017-03-08:
Pikami,

I would suggest that you look around a bit in your /var directory for your missing files.  I am surprised that anything would actually be deleted, but maybe moved or renamed somehow.  I had a little trouble with the install.sh script and my apache2 setup because Apache had made a recent change to its defaults regarding documentroot naming conventions and locations that the install script had not quite caught up to.  But it was nothing I was unable to figure out and fix.   At first, things weren't where I expected them to be, but I found them.  Unfortunately, I am inexperienced with nginx configuration issues.

Take a deep breath and don't give up yet.

-- Fred P.

roberttidey, 2017-03-09:
Fred,
Can you tell me the Apache change you needed to make?

pikami,
I will happily investigate issues reported in a reasonable manner preferably with some details of the event. The installer does not deliberately remove files. If you change install folder then it tries to move previous material to the new folder. Similarly if you uninstall then it tries to move the media folder to the users home directory.

Of course bugs and unforeseen user activity can always cause unexpected behaviour which is why the wise man maintains back ups of any critical information particularly when about to embark on any reconfiguration.

pishotta, 2017-03-09:
Robert,

Looking back on my particular difficulty, I now think the problem was my misunderstanding of what the install script was actually going to do.  I thought rpicamdir was just a destination directory and I would link it up afterwards.  I did not realize that changes to the virtualhost mechanism would also be effected.  (Although it is a little compact in there, I should have examined the script's fn_apache function more closely before I ran it.  Mea culpa!)  Didn't want the Cam interface files mixed in with my other preexisting web files in /var/www/html, so I answered "rpicam" to the prompt "Cam subfolder".  Now realize i probably should have entered "html/rpicam".  Anyway, the DocumentRoot of the * virtualhost got changed to /var/www/rpicam, which is not what I was expecting, so I didn't see my old files via a browser.  They were still there, but under /var/www/html.  No big deal, though.  I manually moved the rpicam folder to under /var/www/html as a subsidiary and edited the pertinent Apache config files appropriately.  That is how I recall the incident -- another brief tale for my memoirs.

Like I said, I think it was my expectations that were incorrect (true too often).  I now think the script did what it was intended to do, given my responses to the prompts.  I just hadn't realized at the time exactly what that was going to be.  With the myriad of possible preexisting web server configurations with which it may have to contend, it's pretty impressive that it works as well as it does.

Hopefully it is something like that that happened to Pikami.

-- Fred P.

pikami, 2017-03-10:
I looked everywhere, my files are gone.
The entire html dir is empty

roberttidey, 2017-03-10:
Have you looked outside the html directory? Are there any other folders under /var/www? Also check to see if there is a media folder in your user home directory (e.g. /home/pi/media).

Do you remember what you were doing when this happened? E.g. updating, re-installing to another folder, removing.

pikami, 2017-03-10:
There is a "/home/pi/media/media" but it's empty.
The only folder in the directory /var/www is html

The files have been deleted during the installation.
During the installation the program asked me for the install directory, the default was "html", I changed it to "webcam"

roberttidey, 2017-03-10:
When installing to a different folder than a previous installation the the install script tries to move the material in the old folder to the new one. As in
`if [ ! "$rpicamdir" == "$rpicamdirold" ]; then
   if [ -e /var/www$rpicamdirold/index.php ]; then
      sudo mv /var/www$rpicamdirold/* /var/www$rpicamdir
   fi
fi
`
Last time I checked this it was working OK, and if the move fails then the material should have been left where it was. I'll run some more tests to see what might cause this to fail.

I don't think this has anything to do with nginx or Apache.

Lewiscowles1986, 2017-09-22:
Ignoring the rude comments, this is a problem, I'm happy to code up a fix for. Instead of having configurable dir and nuking config's etc, I'm sure an implementation can be made to install to `/var/www/vhosts/rpicam` and just removing other configs from `/etc/nginx/sites-enabled` (which should mean they are available in `/etc/nginx/sites-available`

Lighttpd and apache may be additional cases, but I think this helps to modularise the code some more.

roberttidey, 2017-09-22:
I'm not sure I fully understand what you are suggesting on this one but I welcome any contributions. The nginx part of the install was originally contributed by another user and I have just tried to make any obvious changes.

I thought the particular issue here had been that the media folder had got trashed when the web install folder was changed during a subsequent install. As I said earlier I didn't see where this was happening and there is nothing specific in the nginx install function.

The ability to choose a sub-folder during the install is fairly popular and I would be reluctant to lose that piece.

Lewiscowles1986, 2017-09-22:
@roberttidey I was thinking of something so users don't have to choose a folder. What are the chances someone is using `/var/www/vhosts/rpicam$ver` specifically? 

where $ver is the version string appended to the folder name, it's a similar pattern used in linux shared libraries 

That could probably fix most issues for nginx users, and it can be symlinked to `./rpicam`

It also means you get the benefits of people choosing a folder (lack of collisions) with simpler time in resolving issues.

It's your repo, so I'll let you tell me the ground rules

closed 2019-10-01 by pikami