silvanmelchior/RPi_Cam_Web_Interface #334: How to uninstall the RPi_Cam_Web_Interface_ServoBlaster_pan_tilt ?

Pilou72, 2017-03-14:
I installed this repositery, and it didn't work, because the installation is made in the wrong folder of apache :
https://github.com/skalad/RPi_Cam_Web_Interface_ServoBlaster_pan_tilt

There is no uninstall.sh, could you tell me how to remove it please? or even better, how to install it in the Html folder of Apache, if you think it still possible to make it work.

Thks a lot,
Pierre

Flux78, 2017-03-15:
Did you try the remove.sh inside the RPi_Cam_Web_Interface folder?
Btw: If you run update.sh with a changed web path, the script will move your old installation inside the new one.
I hope this will help you.

Pilou72, 2017-03-15:
Actually, there isn't any remove.sh or update.sh (or uninstall.sh) in this "forked" repositery : https://github.com/skalad/RPi_Cam_Web_Interface_ServoBlaster_pan_tilt

The only script executable in the folder is : RPi_Cam_Web_Interface_Installer.sh

Is it right to apply a "rm -Rf" on the folder cloned from Github "RPi_Cam_Web_Interface_ServoBlaster_pan_tilt" to remove it all, 
or should I clean some other files or folders ?

Thks a lot,
Pierre

roberttidey, 2017-03-15:
That will remove the installation folder which you want to do. There will also be the runtime folder which will be located under /var/www , either in that folder or in a subfolder under it depending on where you chose during installation. You want to remove that as well using the same type of command but you will need to add sudo on the front. Note that if you have any recordings you want to keep they will be in the media folder there, so save them if you want them.

I'd then recommend using the standard repository for future installs as it is very easy to activate pan tilt support in there anyway.

Pilou72, 2017-03-15:
Thanks !!
 When I get Servoblaster to work, I'll try to activate pan tilt...
For now, I'm stuck with an error, when I do sudo ./servod, to set up the program, I have an error message : "Servod: cannot parse the hardware name string"
I had it working, a week ago... Grrrr...