silvanmelchior/RPi_Cam_Web_Interface #336: motion detection stops   from time to time

shinji2009, 2017-03-15:
hi. motion detection stops  for no reason from time to time. example in log after 07:07:45. Motion detect mode: internal. md 1 in Period Start. if press motion detection start button it starts again. last recorded file become "busy" forever. raspbian jessie lite. 2 x rpi3b. plenty of space on a 32gb microsd - min 2gb purge files every 5 mins. help please. how to find why and how to fix.

Code: Select all
{2017/03/13 07:07:44} recommended video buffers 1
{2017/03/13 07:07:44} h264 buffers set to recommended 1
{2017/03/13 07:07:44} Set up internal detect width=121 height=68
{2017/03/13 07:07:44} Opening FIFO 0 /var/www/html/FIFO 7
{2017/03/13 07:07:44} Opening FIFO 1 /var/www/html/FIFO11 8
{2017/03/13 07:07:44} MJPEG streaming, ready to receive commands
{2017/03/13 07:07:44} send smd 9
{2017/03/13 07:07:44} Starting command loop
[2017/03/13 07:07:44] Reload parameters command requested
[2017/03/13 07:07:44] Scheduler loop is started
{2017/03/13 07:07:45} DEBUG 3
[2017/03/13 07:07:45] Scheduled management tasks. Next at 1489363965
[2017/03/13 07:12:46] Scheduled management tasks. Next at 1489364266
[2017/03/13 07:17:47] Scheduled management tasks. Next at 1489364567
[2017/03/13 07:22:48] Scheduled management tasks. Next at 1489364868
[2017/03/13 07:27:49] Scheduled management tasks. Next at 1489365169
[2017/03/13 07:32:50] Scheduled management tasks. Next at 1489365470

roberttidey, 2017-03-15:
That bit of log is like a restart. Is there anything before the 'recommended video buffers' entry?

What mode do you have scheduler set to (All Day, Sun based, Fixed Times)?

A recording that says busy forever means that the mp4 file doesn't exist indicating that it was not MP4Boxed into a mp4 file. The log should show that processing being done.

shinji2009, 2017-03-15:
sorry i not save that log. here's a new one: (18:24:50 - manual start)
`[2017/03/15 17:13:32] Start capture requested from Pipe
[2017/03/15 17:13:32] Send ca 1
{2017/03/15 17:13:32} Capturing with split of 300 seconds
{2017/03/15 17:13:32} Capturing started
{2017/03/15 17:13:32} Finished boxing /var/www/html/media/vi_3170_20170315_171242.mp4 from Box Queue at pos 0
{2017/03/15 17:13:32} Removed item from Box Queue
{2017/03/15 17:13:52} send smd 0
[2017/03/15 17:13:52] Stop capture requested
[2017/03/15 17:13:52] Send ca 0
{2017/03/15 17:13:52} Capturing stopped
{2017/03/15 17:13:52} Add /var/www/html/media/vi_3171_20170315_171332.mp4 to Box Queue at pos 1
{2017/03/15 17:13:52} DEBUG 3
{2017/03/15 17:13:52} Start boxing /var/www/html/media/vi_3171_20170315_171332.mp4.h264 to /var/www/html/media/vi_3171_20170315_171332.mp4 Queue pos 1
{2017/03/15 17:13:53} Finished boxing /var/www/html/media/vi_3171_20170315_171332.mp4 from Box Queue at pos 1
{2017/03/15 17:13:53} Removed item from Box Queue
{2017/03/15 17:14:21} send smd 1
[2017/03/15 17:14:21] Start capture requested from Pipe
[2017/03/15 17:14:21] Send ca 1
{2017/03/15 17:14:21} Capturing with split of 300 seconds
{2017/03/15 17:14:21} Capturing started
{2017/03/15 17:14:36} send smd 0
[2017/03/15 17:14:36] Stop capture requested
[2017/03/15 17:14:36] Send ca 0
{2017/03/15 17:14:36} Capturing stopped
{2017/03/15 17:14:36} Add /var/www/html/media/vi_3172_20170315_171421.mp4 to Box Queue at pos 2
{2017/03/15 17:14:36} DEBUG 3
{2017/03/15 17:14:37} Start boxing /var/www/html/media/vi_3172_20170315_171421.mp4.h264 to /var/www/html/media/vi_3172_20170315_171421.mp4 Queue pos 2
{2017/03/15 17:14:38} Finished boxing /var/www/html/media/vi_3172_20170315_171421.mp4 from Box Queue at pos 2
{2017/03/15 17:14:38} Removed item from Box Queue
{2017/03/15 17:14:45} send smd 1
[2017/03/15 17:14:45] Start capture requested from Pipe
[2017/03/15 17:14:45] Send ca 1
{2017/03/15 17:14:45} Capturing with split of 300 seconds
{2017/03/15 17:14:45} Capturing started
{2017/03/15 17:15:17} send smd 0
[2017/03/15 17:15:17] Stop capture requested
[2017/03/15 17:15:17] Send ca 0
{2017/03/15 17:15:17} Capturing stopped
{2017/03/15 17:15:17} Add /var/www/html/media/vi_3173_20170315_171445.mp4 to Box Queue at pos 3
{2017/03/15 17:15:17} DEBUG 3
{2017/03/15 17:15:18} send smd 1
{2017/03/15 17:15:18} Start boxing /var/www/html/media/vi_3173_20170315_171445.mp4.h264 to /var/www/html/media/vi_3173_20170315_171445.mp4 Queue pos 3
{2017/03/15 17:15:19} Finished boxing /var/www/html/media/vi_3173_20170315_171445.mp4 from Box Queue at pos 3
{2017/03/15 17:15:19} Removed item from Box Queue
[2017/03/15 17:15:19] Start capture requested from Pipe
[2017/03/15 17:15:19] Send ca 1
{2017/03/15 17:15:19} Capturing with split of 300 seconds
{2017/03/15 17:15:19} Capturing started
[2017/03/15 17:15:23] Scheduled management tasks. Next at 1489573223
[2017/03/15 17:15:23] Purged 6 Files
{2017/03/15 17:15:58} Watchdog detected problem. Stopping{2017/03/15 17:16:01} SIGINT/SIGTERM received, stopping
{2017/03/15 17:16:04} Capturing stopped
{2017/03/15 17:16:04} Add /var/www/html/media/vi_3174_20170315_171519.mp4 to Box Queue at pos 4
{2017/03/15 17:16:10} RaspiMJPEG Version 5.8.01
{2017/03/15 17:16:10} recommended video buffer size 65536
{2017/03/15 17:16:10} h264 size set to 131072
{2017/03/15 17:16:10} recommended video buffers 1
{2017/03/15 17:16:10} h264 buffers set to recommended 1
{2017/03/15 17:16:10} Set up internal detect width=121 height=68
{2017/03/15 17:16:10} Opening FIFO 0 /var/www/html/FIFO 7
{2017/03/15 17:16:10} Opening FIFO 1 /var/www/html/FIFO11 8
{2017/03/15 17:16:10} MJPEG streaming, ready to receive commands
{2017/03/15 17:16:10} send smd 9
{2017/03/15 17:16:10} Starting command loop
[2017/03/15 17:16:10] Reload parameters command requested
[2017/03/15 17:16:10] Scheduler loop is started
{2017/03/15 17:16:10} DEBUG 3
[2017/03/15 17:16:11] Scheduled management tasks. Next at 1489573271
[2017/03/15 17:16:11] Purged 1 Files
[... 70 lines cut ...]
[2017/03/15 11:17:04] Send ca 1
{2017/03/15 11:17:04} Capturing with split of 300 seconds
{2017/03/15 11:17:04} Error: Could not send buffers to port, h264 callback
{2017/03/15 11:17:04} Executing macro /var/www/html/macros/error_hard.sh "Could not send buffers to port, h264 callback"
{2017/03/15 11:17:09} RaspiMJPEG Version 5.8.01
{2017/03/15 11:17:09} recommended video buffer size 65536
{2017/03/15 11:17:09} h264 size set to 131072
{2017/03/15 11:17:09} recommended video buffers 1
{2017/03/15 11:17:09} h264 buffers set to recommended 1
{2017/03/15 11:17:09} Set up internal detect width=121 height=68
{2017/03/15 11:17:09} Opening FIFO 0 /var/www/html/FIFO 7
{2017/03/15 11:17:09} Opening FIFO 1 /var/www/html/FIFO11 8
{2017/03/15 11:17:09} MJPEG streaming, ready to receive commands
{2017/03/15 11:17:09} send smd 9
{2017/03/15 11:17:09} Starting command loop
[2017/03/15 11:17:10] Reload parameters command requested
[2017/03/15 11:17:10] Scheduler loop is started
[2017/03/15 11:17:11] Scheduled management tasks. Next at 1489551731
[2017/03/15 11:17:11] Purged 5 Files
[2017/03/15 11:22:12] Scheduled management tasks. Next at 1489552032
[2017/03/15 11:27:13] Scheduled management tasks. Next at 1489552333
...
[2017/03/15 18:18:35] Scheduled management tasks. Next at 1489577015
[2017/03/15 18:23:36] Scheduled management tasks. Next at 1489577316
{2017/03/15 18:26:24} Internal motion detection started`

roberttidey, 2017-03-15:
In first file 17:15:58} Watchdog detected problem. Stopping{2017/03/15 17:16:01} SIGINT/SIGTERM received, stopping

This indicates the watchdog detected that the camera had stopped producing preview images and restarted the system. The file currently being captured would not be processed through to an mp4

In the second example 11:17:04} Error: Could not send buffers to port, h264 callback
This also indicates a problem with the camera subsystem which led to the software restarting.

I'll check further into what might trigger this problem. A few things that I have seen before causing this type of issue are

marginal power supply: The camera introduces an extra power load. It is not just the power supply itself that should be OK. It is also the power/usb cable which needs to be low resistance. Many usb cables have a high resistance (> 0.3 Ohm) and cause significant voltage drop.

camera connector: sometimes the tiny ribbon on the camera module itself can become loose.  Press down to ensure good connection

gpu memory. Normally this is set to 128MB which is OK most of the time. If less then it should definitely be increased to 128M. It may also be worth trying to increase it a bit (e.g. 144MB) to see if that eliminates the problem.

shinji2009, 2017-03-15:
thank you for answer
can you please tell how to change gpu memory? i can't find any info about it. sorry i am newbie at all this

 about power: i was watch for boards for a long time and not seen any problem with power supply (red led not flashing), i'm using single 5v 3.6a power supply for both boards and no usb devices are connected. i was using another surveillance software for a couple of weeks with this configuration and cameras was not hanging... or was restarted in more hard way) so i wasn't notice they hang or big gaps in videos. ribbon cables connectors  also so tight i was able to close them only by two hands... but i'll reattach them and even clean ribbon pins using eraser

shinji2009, 2017-03-15:
eraser not help

btw why motion detection don't turn on automatically after this? everything seem working fine for a while after press motion detection start button. is md 1 in Period Start not enough?

roberttidey, 2017-03-15:
I think there is a case when new scheduling period doesn't get detected after a camera restart so md 1 doesn't get sent. I think I have fixed that in 6.3.2

I am still checking why it may be failing for you. I have attached here an earlier version of the raspimjpeg process built before some of the recent changes. It would be good if you could try using this to see if it solves your problem. I can then focus on what change could be the problem.

To put this in
Stop camera system by running ./stop.sh in the install folder
Get raspimjpeg out of the zip and put in your home directory
sudo cp ~/raspimjpeg /opt/vc/bin/raspimjpeg
Start the camera system by running ./start.sh in the install folder.

[raspimjpeg.zip](https://github.com/silvanmelchior/RPi_Cam_Web_Interface/files/845521/raspimjpeg.zip)

shinji2009, 2017-03-16:
not help? 144mb also not help

{2017/03/16 19:25:23} Add /var/www/html/media/vi_3880_20170316_192501.mp4 to Box Queue at pos 17
{2017/03/16 19:25:23} DEBUG 3
{2017/03/16 19:25:23} Start boxing /var/www/html/media/vi_3880_20170316_192501.mp4.h264 to /var/www/html/media/vi_3880_20170316_192501.mp4 Queue pos 17
{2017/03/16 19:25:24} Finished boxing /var/www/html/media/vi_3880_20170316_192501.mp4 from Box Queue at pos 17
{2017/03/16 19:25:24} Removed item from Box Queue
{2017/03/16 19:26:20} send smd 1
[2017/03/16 19:26:20] Start capture requested from Pipe
[2017/03/16 19:26:20] Send ca 1
{2017/03/16 19:26:20} Capturing started
{2017/03/16 19:27:02} send smd 0
[2017/03/16 19:27:02] Stop capture requested
[2017/03/16 19:27:02] Send ca 0
{2017/03/16 19:27:02} Capturing stopped
{2017/03/16 19:27:02} Add /var/www/html/media/vi_3881_20170316_192620.mp4 to Box Queue at pos 18
{2017/03/16 19:27:02} DEBUG 3
{2017/03/16 19:27:02} Start boxing /var/www/html/media/vi_3881_20170316_192620.mp4.h264 to /var/www/html/media/vi_3881_20170316_192620.mp4 Queue pos 18
{2017/03/16 19:27:08} send smd 1
{2017/03/16 19:27:20} Watchdog detected problem. Stopping[2017/03/16 19:27:21] Start capture requested from Pipe
[2017/03/16 19:27:21] Send ca 1
{2017/03/16 19:27:21} SIGINT/SIGTERM received, stopping
{2017/03/16 19:27:30} RaspiMJPEG Version 5.7.06
{2017/03/16 19:27:30} recommended video buffer size 65536
{2017/03/16 19:27:30} h264 size set to 131072
{2017/03/16 19:27:30} recommended video buffers 1
{2017/03/16 19:27:30} h264 buffers set to recommended 1
{2017/03/16 19:27:30} Set up internal detect width=121 height=68
{2017/03/16 19:27:30} Opening FIFO 0 /var/www/html/FIFO 7
{2017/03/16 19:27:30} Opening FIFO 1 /var/www/html/FIFO11 8
{2017/03/16 19:27:30} MJPEG streaming, ready to receive commands
{2017/03/16 19:27:30} send smd 9
{2017/03/16 19:27:30} Starting command loop
[2017/03/16 19:27:30] Reload parameters command requested
[2017/03/16 19:27:30] Scheduler loop is started
{2017/03/16 19:27:30} DEBUG 3
[2017/03/16 19:27:31] Scheduled management tasks. Next at 1489667551
[2017/03/16 19:27:31] Purged 2 Files
[2017/03/16 19:32:32] Scheduled management tasks. Next at 1489667852
[2017/03/16 19:37:33] Scheduled management tasks. Next at 1489668153
[2017/03/16 19:42:34] Scheduled management tasks. Next at 1489668454
[2017/03/16 19:47:35] Scheduled management tasks. Next at 1489668755
[2017/03/16 19:52:36] Scheduled management tasks. Next at 1489669056
[2017/03/16 19:57:37] Scheduled management tasks. Next at 1489669357

shinji2009, 2017-03-26:
/etc/raspimjpeg motion_detection true do the trick
but i still getting same messages in the log and get permanently busy files often
they are not downloadable, get a 0 bytes zip file

shinji2009, 2017-03-27:
no, it was a early rejoice
moreover interface was stuck at 19:27:58 at recording video... and preview freezes too(

[2017/03/27 19:26:44] Send ca 0
{2017/03/27 19:26:44} Capturing stopped
{2017/03/27 19:26:44} Add /var/www/html/media/vi_13189_20170327_192309.mp4 to Box Queue at pos 4
{2017/03/27 19:26:45} Start boxing /var/www/html/media/vi_13189_20170327_192309.mp4.h264 to /var/www/html/media/vi_13189_20170327_192309.mp4 Queue pos 4
{2017/03/27 19:27:06} send smd 1
{2017/03/27 19:27:19} Watchdog detected problem. Stopping[2017/03/27 19:27:22] Start capture requested from Pipe
[2017/03/27 19:27:22] Send ca 1
{2017/03/27 19:27:22} SIGINT/SIGTERM received, stopping
{2017/03/27 19:27:34} RaspiMJPEG Version 5.8.01
{2017/03/27 19:27:39} recommended video buffer size 65536
{2017/03/27 19:27:39} h264 size set to 131072
{2017/03/27 19:27:39} recommended video buffers 1
{2017/03/27 19:27:39} h264 buffers set to recommended 1
{2017/03/27 19:27:39} send smd 1
{2017/03/27 19:27:39} Set up internal detect width=121 height=68
{2017/03/27 19:27:39} Opening FIFO 0 /var/www/html/FIFO 7
{2017/03/27 19:27:39} Opening FIFO 1 /var/www/html/FIFO11 8
{2017/03/27 19:27:39} MJPEG streaming, ready to receive commands
{2017/03/27 19:27:39} send smd 9
{2017/03/27 19:27:39} Starting command loop
[2017/03/27 19:27:39] Start capture requested from Pipe
[2017/03/27 19:27:39] Send ca 1
{2017/03/27 19:27:40} send smd 1
{2017/03/27 19:27:40} Capturing with split of 300 seconds
{2017/03/27 19:27:40} Capturing started
[2017/03/27 19:27:42] Reload parameters command requested
[2017/03/27 19:27:42] Scheduler loop is started
[2017/03/27 19:27:42] Start capture request ignored, day period not initialised yet
[2017/03/27 19:27:43] Scheduled management tasks. Next at 1490617963
{2017/03/27 19:27:58} send smd 0
{2017/03/27 19:28:08} Watchdog detected problem. Stopping{2017/03/27 19:28:17} SIGINT/SIGTERM received, stopping
[2017/03/27 19:28:19] Purged 27 Files
[2017/03/27 19:28:33] Stop capture requested
[2017/03/27 19:28:33] Send ca 0
[2017/03/27 19:28:36] New period detected 0
[2017/03/27 19:28:36] Send md 1
[2017/03/27 19:32:44] Scheduled management tasks. Next at 1490618264
[2017/03/27 19:37:45] Scheduled management tasks. Next at 1490618565
[2017/03/27 19:42:46] Scheduled management tasks. Next at 1490618866
[2017/03/27 19:47:47] Scheduled management tasks. Next at 1490619167
[2017/03/27 19:52:48] Scheduled management tasks. Next at 1490619468
[2017/03/27 19:57:49] Scheduled management tasks. Next at 1490619769
[2017/03/27 20:02:50] Scheduled management tasks. Next at 1490620070
[2017/03/27 20:07:51] Scheduled management tasks. Next at 1490620371
[2017/03/27 20:12:52] Scheduled management tasks. Next at 1490620672
[2017/03/27 20:17:53] Scheduled management tasks. Next at 1490620973

shinji2009, 2017-05-16:
maybe it will be a good idea to give to a watchdog more power? for example let it reboot the device if it detects something fishy. yesterday one of my boards catch this error {2017/05/15 17:46:08} Error: Could not enable connection camera -> video converter, after that RPi_Cam_Web_Interface act like everything is ok, webinterface preview was working, indicate that it is detect and recording something, but actually it was only shedluied purges in log after that, nothing was recorded

roberttidey, 2017-05-16:
Yes. That is quite a strange error as it indicates that just part of the MMAL interface has broken down. If the preview continues then the watchdog thinks the system is OK.

However, errors are handled by a central routine. They are classified as fatal or non-fatal. Fatal ones like any of the MMAL interface errors including that one try to call the error_hard.sh macro if present in the macros folder and then exit the current thread. The watchdog should kick in but it may be possible for the preview callbacks to keep it alive. Non fatal errors try to call error_soft.sh macro before continuing normal operation.

The attempt to call the error_hard.sh macro should be in the schedule log. A simple macro could reboot the Pi under these circumstances.

metalshreds, 2017-05-30:
I'm having the same issue, I'm using this as a security camera facing my front door. After a few hours I come back to it and the motion capture has stopped. Possibly unrelated but thought I'd mention, even though purge_level is set to Min % set to 80%, when I come back to it, all videos are gone and motion has stopped. I'm not sure how to pull logs but let me know if you need any. (Rpi3, 16GB sd)

roberttidey, 2017-05-30:
Min space % is the minimum space relative to the overall capacity of the drive that the purge level is trying to maintain. If you have that set to 80% then the purge is trying to keep 80% of 16GB free. That is 12.8GB free. The overall OS and software is probably taking more than the 3.2GB so the purge ends up removing everything in an effort to meet the 80%. 

You probably meant that you didn't want to exceed 80% overall. In that case set the minimum space to 20% or use one of the other purge settings like Max usage. 

With your original setting it is also possible that the purge is trying to remove live recordings whilst they are happenning with unpredictable results.

shinji2009, 2017-05-31:
roberttidey,  is it possible to make this error fatal so it reboot system or something so recording and motion detection actually don't stop?

metalshreds, 2017-06-01:
Thanks for the clarification, I read the sentence on the wiki multiple times but found it confusing :/ . Hopefully this will resolve my problems! thanks!

tman32768, 2017-06-13:
I have the same issue.  It happens most frequently using internal motion detection on sunny and windy days where the camera picks up a lot of false positives for motion due to tree leaves and ground shadows moving in the images.  I also use a mask file but it doesn't appear to work.

giuaig, 2017-08-17:
I'm experiencing the same issue too. I'm using full-day schedule, internal detection and last 6.3.14 version.
I attach the log here in the hope that this can help hunting the bug:
[20170817-101856-scheduleLog-01.txt](https://github.com/silvanmelchior/RPi_Cam_Web_Interface/files/1230795/20170817-101856-scheduleLog-01.txt)

shinji2009, 2018-03-18:
hi. why /etc/raspimjpeg motion_detection  is not working anymore? i have installed RPi Cam Web Interface anew and motion detection alwasys stops after a while

roberttidey, 2018-03-18:
You need to supply more info than that. Motion detection on my cameras does not stop after years of operation.

1) Which version you installed
2) Which Raspbian Date / Full or Lite
3) Using internal or external motion
4) Schedule set up
5) Schedule log when it stops

shinji2009, 2018-03-21:
it was Scheduler. sorry. i was thinking that /etc/raspimjpeg motion_detection true is stronger than Scheduler.