silvanmelchior/RPi_Cam_Web_Interface #337: Day by day scheduler

ebretteville, 2017-03-20:
Hye,
Just get the scheduler working. It was my fault, but at a first try, the syntax and scheduler has it's logic & complexity ;-) One thing that I can not achieve and maybe this is the way it should behave is that I can't specify some period by day.
Is it hidden somewhere? Do you plan to make this available?

Thanks for your hard work!

roberttidey, 2017-03-20:
The scheduler currently has the 3 modes for setting schedule periods.
All Day - which just means active all the time
Sun based which has 4 periods based on calculated sun rise and sun set times and which adjust according to location and date
Fixed Times which allows 6 periods which are the same every day.

I think you are asking for Fixed Times but the times to be varied according to the day of the week. Is that right?

The scheduler currently doesn't support that. Altering the scheduler processing to do that wouldn't be too difficult to do. But one would need to think about the user interface part to allow that to be defined without complicating it for the more normally case of same times every day.

ebretteville, 2017-03-20:
> I think you are asking for Fixed Times but the times to be varied according to the day of the week. Is that right?

Yes that's what I mean. Changing the period by day or workday etc.

Yes it's true it has to be simple and not to make things harder than existing setup

ebretteville, 2017-03-21:
Maybe for each period raw, a new column with 7 tick boxes to select which day this period is activated?

Just my 2cent

roberttidey, 2017-03-27:
This is included in latest version

ebretteville, 2017-03-27:
Thanks so much.

closed 2017-03-27 by ebretteville