silvanmelchior/RPi_Cam_Web_Interface #338: Cam image in the Raspberry remote?

rogercorrea, 2017-03-21:
Please,

It is possible install the web interface in a Raspberry and visualize the image camera installed in other Raspberry in the same network?

I have three boards with the RPi cam and I don't know if it is possible.
Thanks!

rogercorrea, 2017-03-23:
Do anyone help me?

roberttidey, 2017-03-23:
The web interface should be accessible from most web browsers.  So if you browse from one raspberry to another supporting the camera interface then it should be viewable.

rogercorrea, 2017-03-23:
Thanks, @roberttidey !
But I need show the thumbnails with all the interfaces in the same page.
I think that I can change the code to work as I need. I can mount the code made to one interface to show many interfaces in the same page, correct?

roberttidey, 2017-03-23:
Yes. It should be fairly straightforward to construct a page that showed multiple views from different cameras.

You would need to have a home page on one of the servers that knew the IP addresses of each of the cameras.

An advanced version would allow drilling through to access the settings and captures on each server.

If you do any work on this then please let me know so I can make it available for general use.