silvanmelchior/RPi_Cam_Web_Interface #340: Option to only use gray-scale

theborax, 2017-03-29:
Hi,

In the look of making things more bandwidth-efficient I've been looking around on how to only use gray-scale in this product, but to no avail. Just taking a still picture and opening it in a picture editor and making it gray-scale makes the data about a 1/3 of the size. For me resolution would be more beneficial over color (using a no-ir camera so the colors are skewed anyway). So my request is to have a "pure gray-scale" option somewhere.

Thanks for an otherwise good project :).

roberttidey, 2017-03-29:
Just to clarify where are you aiming to save bandwidth? In the live preview or in viewing the captures? These use separate paths through the camera interface.

theborax, 2017-03-29:
Sorry, both would be interesting but live view would be my primary wish. Just used the still picture mode to get some view on how much file size differs when going grayscale (makes sense that it's 1/3 as 3 color channels goes down to one).

pathammer, 2017-05-26:
This is a good idea. Yes it would reduce bandwidth unfortunately I don't think raspimjpeg supports this. You could obviously set saturation to -100 in the GUI as a simulation of this effect.

roberttidey, 2017-05-26:
I have played around with this a bit but I don't think it is giving much reduction in bandwidth. 

The main reason I think is that the colour encoding is actually very efficient in jpeg and mp4. It is not a simple 3 times reduction. The YUV scheme means that all the detail is in the Y luminance channel and the UV chrominance channels are at /2 resolution (/4 data) so that total raw data is maybe 1.5 times and the compression may reduce this factor further.

I did manage to get about a 15% reduction by gray scaling. 

You may be able to get a bit more by converting to gray as suggested using by enabling colour effects U:V = 128 and lowering the preview quality a little bit. The gray scale nature may allow a slightly lower quality to be used.