silvanmelchior/RPi_Cam_Web_Interface #341: 03-02-17 Jessie Lite build Install issues

mzoebelein, 2017-04-10:
I had issues with the Apache installer on the most recent version of Jessie Lite (3-02-17). I resolved the issue by installing Apache from "RPi_Cam_Web_Interface_Installer.sh". Worked like a charm ofter that, fantastic software!

roberttidey, 2017-04-10:
OK. Thanks for the feedback. Will investigate.

If you have any extra info on the issues encountered that would be helpful.

mzoebelein, 2017-04-10:
Sure, I installed it twice on fresh images and ran into the same issue. It might be a red herring but, I ran into issues with the chmod command and I had to use "chmod a+x" instead of "chmod u+x" after several tries with nothing working. After that I ran ./install.sh and the wizard worked but had errors on the Apache portion of the install. Again this was a fresh image of Jessie Lite on a Pi Zero 1.3. My SD card was formatted using SD Formatter V4.0  on the second install. It is a cheap 16 GB 4 card but is brand new.

roberttidey, 2017-04-10:
I have run through an install starting from a clean image of the latest 2017-03-02-raspbian-jessie-lite 

I used exactly the steps in the wiki (install.sh) and encountered no errors and had a running camera system up.

Few points to note.
1) It is important to do the apt-get update and apt-get dist-upgrade steps
2) For Jessie Lite one needs to install git
3) chmod u+x worked fine. This gives execution rights to the current user normally pi. chmod a+x gives rights to everyone. Is it possible you have created another user who doesn't have the rights that pi has?
4) When install.sh is run then it creates a log file called install.txt. This can be helpful in debugging any install problems. This won't have got overwritten even if you ran the other installer.

mzoebelein, 2017-04-10:
Thank you. I'll redo the install tonight.