silvanmelchior/RPi_Cam_Web_Interface #343: Request: Linking RPi_Cam_Web_Interfaces together

RichShumaker, 2017-04-17:
I run multiple cameras at the same time using the RPi Cam Web Interface software.
I want to trigger all of them at the same time from a single browser.

I would hit the "Record Image" button and that would activate on all the active RPi Cam Web Interface cameras on the network.  I would also use this to start and stop videos as well.

I would love it if more than just the basic start stop commands were possible.  The ability for you to link all the cameras so that one browser controls all of them would be amazing.  Possibly with a preview for all the cameras on a single browser screen.  I know I could use other software to do a multi-view and I currently use vMix to do that with my cameras.  So having a multi-view is not really the point.  The point is the ability to control and trigger multiple cameras across multiple RasPi's using the software at the same time..

Thank you very much for everything you all do the software is brilliant and amazingly awesome.

Rich Shumaker

RichShumaker, 2017-04-17:
To add to this I am looking at a way to use something like Docker with RPi_Cam_Web_Interface to allow an easy to set up, deploy, use & maintain system of multiple cameras.  Currently I am exploring FIFO pipe and I am trying to get all my settings spit back out as a command string in the same format you use to put it in.
Here is just a basic take image command.

`echo 'im' `>/var/www//html/FIFO`

Ideally I could set up my configuration on one RasPi Cam via the Web Interface.  Then I could easily migrate it using SSH to other systems.  Being able to easily spit out the full setup in command code would allow an easy dump to another RasPi via copy and paste.

I have a feeling I will need to create some code now to allow me to spit out all the current settings so they could be copied to another machine.  Probably best to spit it out to a text file and then copy from that file.

Does this already exist and I just need to know where it is to use it?

roberttidey, 2017-04-17:
The default settings are in the /etc/raspimjpeg file and then any changes to these made from the web interface are stored in the uconfig file held in the web run folder (e.g. /var/www/html). The programs actually read the default file followed by the uconfig file so the latter takes precedence.

A user can manually edit the defaults if so wished and then any subsequent settings changes made from the web interface are stored in uconfig.

RichShumaker, 2017-04-17:
Thanks @roberttidey I will look into that.  I guess I could set up the first system the way I want using the RasPi Cam Web Interface then copy the uconfig file from one system to the next.  I was looking at a way to 'inject' the code from one RasPi to another by using the FIFO Pipe via SSH.  That way I might inject the settings from one system to all of them via a group SSH command somehow.
If I wanted a more permanent setup I could use the default file instead of the uconfig which is subject to change once you modify the web interface.

Is the uconfig file 'executable' via SSH FIFO pipe or does it need to be modified to run that way?

roberttidey, 2017-04-18:
The uconfig file is just a text file in the same format as the raspimjpeg config file. They are just read one after another by the raspimjpeg process when it starts up. If it was updated by something else then the raspimjpeg can be halted (ru 0) and restarted (ru 1) which would cause the config files to be read again.

You should be able to do a remote write into the FIFO pipe to execute commands on other raspberry pis.

RichShumaker, 2017-12-12:
> You should be able to do a remote write into the FIFO pipe to execute commands on other raspberry pis.

How would I do that?
I have researched a bit except I couldn't quite make it happen.
Ideally I have another computer that could send commands to all the Pi's via FIFO except that is beyond me still.

Thank you very much for all the help.

Rich Shumaker

roberttidey, 2017-12-12:
There are various ways, e.g. setting up samba shares but the easiest way is to exploit the web interface existing on the raspberries with cameras. This allows you to access a URL to send a command into the pipe.

E,g http://cameraip/cmd_pipe.php?cmd=ca%201

will start a video recording by sending 'ca 1' into the pipe.

Note that camerip needs to include any subfolder you are using (e.g. ip/html)

The %20 is a space character.

In python this would be something like

import urllib2
urllib2.urlopen("http://cameraip/cmd_pipe.php?cmd=ca%201").read()

RichShumaker, 2017-12-13:
Thanks @roberttidey that rocks.  I will try to inject a few commands tonight.
Do you know of any windows methods to send commands to multiple IP addresses at the same or similar time?  You mentioned Python so I will try to create a script with that.

Ideally I could have one Master Page that would feed all the other Pages, each page being a different camera on the network.

roberttidey, 2017-12-13:
You can use curl for windows to provide url access from the command line and then create batch files to send out the urls you want.

If you are aiming to send out the same command to multiple cameras from a web page then it might be worth adapting the multiview facility. This currently provides simple view windows to multiple camera but by changing multiview.html and multiview.php one could add command buttons to it and then have the modified multiview.php send the commands to the hosts list.

RichShumaker, 2017-12-13:
@roberttidey that sounds amazing.  I injected my first commands tonight and they worked great.  I will try Curl next and maybe a batch to see if I can do that.

I will definitely look at adapting the multiview facility in multiview.html/.php I like that idea a lot.  Is there anywhere I can read more about multiview facility and how to work with it.  I didn't read that anywhere yet so I must have missed it.

Thank you again for all the help.

roberttidey, 2017-12-13:
There is a section on wiki about multiview under additions and tricks. 

If you get that working as a basic multiview then it should be pretty simple to enhance to add on global functions.

RichShumaker, 2017-12-13:
Found the Multiview section on the wiki and I will be experimenting today to see what I can create.
I will post my findings and thank you very much again @roberttidey for helping me figure this all out.

RichShumaker, 2017-12-29:
I am starting my testing now and I had a question, how would you add commands to this page.  Ideally I could start / stop recording and take an image.  I am sure once I have those things on the Mutliview I will want more, heheh.
I am also exploring bluetooth and having some type of bluetooth device to start stop recording and also to take pictures.

Thanks again for your help @roberttidey.

roberttidey, 2017-12-30:
Commands in general from the web interface work by using the send_cmd function in the javascript script.js The function accesses the cmd_pipe.php page on the server which then issues the command.

That is normally  in the context of 1 server/ 1 camera.

Here you want the command to be sent to the appropriate camera. The way to do this is to extend the multiview.php so that it can accept command call, look up the appropriate hosts like multiview.php already does and then access the cmd_pipe.php on that host to pass the command on.

So steps needed are.

1) Add one or more buttons to multiview.html that have an action to call a new javascript function (e.g. send_cmdmulti with 2 parameters to identify the host (or all of them if a global action is wanted) and the command required. A hostnumber of say -1 could be used to indicate it is a command for all hosts.

2) Define the new javascript function to script.js which will access the multiview.php passing on the host and command. Something like

function send_cmd (hostNumber ,cmd) {
  ajax_cmd.open("GET","multiview.php?hostnumber=" + hostnumber + "&cmd=" +cmd ,true);
  ajax_cmd.send();
}

3) Modify multiview.php to check if a hostnumber argument is included. If so look up the host address and access its cmd_pipe.php with the cmd parameter. If global then the multiview.php would do the cmd_pipe access on all the hosts in its table one by one.

RichShumaker, 2018-01-17:
I took the multiview.jsonD template and updated it with my 3 cameras, unfortunately I didn't know where to comment out the fourth camera, d'oh!!!
IT WORKS GREAT!!!
Adding a button to multiview seems to be a bit beyond my skill set right now.
I know multiview is the fastest way for me to trigger or start/stop all my cameras.  I don't need the previews although those are very cool and I like them.  Ideally I would just have the IP's of all the cameras on screen or a link to see them to keep the page from getting cluttered. 5 Cameras Online - then you click(5 Cameras Online) and it lists the IP's as links so you could go to them that way.
For the Main Page I would have 2 buttons, 1. Take Picture 2. Start/Stop Video

Thanks again for your help I will need to keep digging on this.  Except I know this will work once I can figure it all out.
I need to learn more quickly that way I can close this out as it has dragged on and that is my fault.

RichShumaker, 2018-01-19:
@roberttidey Is there a place where someone can ask a question about this software BESIDES the RasPi Forum?  That thread is years long literally and I just have a few questions like, how could I use bluetooth to control multiple cameras?  How do I change the presets easily and where for the 2 different cameras?  Actually I have a lot of questions and reading the forum thread is like trolling thru mud.
Thanks again for all the help Multiview is really cool.  Does multiview need a faster CPU or can it run find on any RasPi

XavSmadja, 2022-06-07:
> I took the multiview.jsonD template and updated it with my 3 cameras, unfortunately I didn't know where to comment out the fourth camera, d'oh!!! IT WORKS GREAT!!! Adding a button to multiview seems to be a bit beyond my skill set right now. I know multiview is the fastest way for me to trigger or start/stop all my cameras. I don't need the previews although those are very cool and I like them. Ideally I would just have the IP's of all the cameras on screen or a link to see them to keep the page from getting cluttered. 5 Cameras Online - then you click(5 Cameras Online) and it lists the IP's as links so you could go to them that way. For the Main Multiview Page I would have 2 buttons, 1. Take Picture 2. Start/Stop Video
> 
> Thanks again for your help I will need to keep digging on this. Except I know this will work once I can figure it all out. I need to learn more quickly that way I can close this out as it has dragged on and that is my fault.

Hi @RichShumaker ! I saw that you worked on the Multiview.php file to create a common command for several cameras. You know what? I've been working on it for 4 weeks without success! I still don't understand all the subtleties of the program, and I have only a very small experience of dev, I'm usually rather web...  
[image]

Do you think you could help me? I can't make my buttons work, they don't have any value (ex record video start). Could you send me some files that you have modified? I don't think they are in your github directory! You would save my life!

XavSmadja, 2022-06-15:
> Les commandes en général de l'interface Web fonctionnent en utilisant la fonction send_cmd dans le script javascript.js La fonction accède à la page cmd_pipe.php sur le serveur qui émet ensuite la commande.
> 
> C'est normalement dans le contexte de 1 serveur/1 caméra.
> 
> Ici, vous voulez que la commande soit envoyée à la caméra appropriée. La façon de le faire est d'étendre le multiview.php afin qu'il puisse accepter l'appel de commande, rechercher les hôtes appropriés comme multiview.php le fait déjà, puis accéder au cmd_pipe.php sur cet hôte pour transmettre la commande.
> 
> Donc, les étapes nécessaires sont.
> 
> 1. Ajoutez un ou plusieurs boutons à multiview.html qui ont une action pour appeler une nouvelle fonction javascript (par exemple send_cmdmulti avec 2 paramètres pour identifier l'hôte (ou tous si une action globale est souhaitée) et la commande requise. Un numéro d'hôte de dire -1 peut être utilisé pour indiquer qu'il s'agit d'une commande pour tous les hôtes.
> 2. Définissez la nouvelle fonction javascript à script.js qui accédera au multiview.php en passant l'hôte et la commande. Quelque chose comme
> 
> function send_cmdmulti (hostNumber ,cmd) { ajax_cmd.open("GET","multiview.php?hostnumber=" + hostnumber + "&cmd=" +cmd ,true); ajax_cmd.send(); }
> 
> 3. Modifiez multiview.php pour vérifier si un argument hostnumber est inclus. Si c'est le cas, recherchez l'adresse de l'hôte et accédez à son cmd_pipe.php avec le paramètre cmd. S'il est global, le multiview.php ferait l'accès cmd_pipe sur tous les hôtes de sa table un par un.

Hi @roberttidey are you still here? I would like to go deeper into your great project, and i have a lot of difficulties with the steps you described here! 

Thanks for your answer that would be so cool!