silvanmelchior/RPi_Cam_Web_Interface #346: How to add configuration

royord, 2017-04-24:
It's late, and I've been working this for a number of hours...

I'm trying to figure out how to add some additional options needed for my install... I would like to add things link SCP to the configuration webpage and I'm looking to capture some text fields and an SSH public key. I'm having issues saving these items into the "uconfig" file. I have traced down how the command is executed, but it seems as though the commands are being "aliased" where do I configure the commands that are issued and how they effect the "uconfig" stored text/configuration.

roberttidey, 2017-04-24:
The underlying camera support process raspimjpeg uses the /etc/raspimjpeg config file together with the uconfig file. They have the same format but the uConfig file only stores values that the user has changed in the web interface. Whenever raspimjpeg loads config values, it loads the defaults followed by the uconfig file so the latter effectively applies the changes.

It sounds like you are trying to add functionality that can could be handled by the web side alone rather than involving raspimjpeg. If that is the case and you are modifying the php server side alone then it may be best just to use a separate config file for that purpose, This is what the scheduler does; saving its settings in schedule.json

If you are wanting to embed different functionality in the raspimjpeg process itself and add extra pipe commands then the source for this is listed in the wiki

royord, 2017-04-24:
@roberttidey thanks for your feedback, I have added this to the file "/etc/raspimjpeg/raspimjpeg.1" and have a default that is coming in, it's just not saving when I make the update. I think that this is due to the fact that things such as "send_cmd('an ' + encodeURI(document.getElementById('annotation').value))" is mapped somehow to updating "uconfig" outside of the code that I have been able to find to this point.

I wanted to make sure that I fit my code into the current standard instead of adding my own configuration file and gave up last night. I'll look at the possibility of adding my own config file to handle it.

I'm trying to add dialup for a USB modem and ftp for a location that requires this type of connectivity. Trying to add things such as phone/user/pass. FTP(SCP)/user/pass.

roberttidey, 2017-04-25:
The raspimjpeg.1 file is only used when installing. It is copied to /etc/raspimjpeg which is then the operational version. When one is modifying the code and wants a new parameter then it is appropriate to put a new default value in the raspimjpeg.1 file so it becomes available on new installations.

uconfig gets updated normally by the raspimjpeg process which is handling most of the commands.

So send_cmd triggers the server to write the command to the FIFO pipe. raspimjpeg is reading the pipe and actions the command. When this involves the configuration then raspimjpeg also updates the uconfig file appropriately. In the raspimjpeg source code (https://github.com/roberttidey/userland) the file host_applications/linux/apps/raspicam/RaspiMCmds.c does the command parsing and processing. functions in RaspiMjpeg.c are responsible for changing and saving values in the uconfig file.

It is possible to put new stuff in raspimjpeg config file as unknown stuff is just ignored. The uconfig file though only contains key/values that are known to the raspimjpeg process and different from the default ones in /etc/raspimjpeg process.

So if the changes you are contemplating involve adding new Pipe commands and changing the raspimjpeg process to support them then it is appropriate to put them in the raspimjpeg process and maintain them in uconfig. This can be fairly simple to do as I have organised the code so that it s relatively easy to add new config keys in, but it does mean compiling / building raspimjpeg.

royord, 2017-04-25:
@roberttidey Thanks a bunch, none of this actually scares me, I'll fork it here locally and push back changes when I have something that is workable.

royord, 2017-04-25:
@roberttidey any reason that you point to your userland and not the one on this repo? Your repo for userland hasn't been recently updated compared to the userland on this repo. I'm going to look at using the one that raspberry keeps unless otherwise noted.

roberttidey, 2017-04-25:
my userland is a fork of the raspberry one to provide a build environment for the raspimjpeg process used in the web app. There is no raspimjpeg process in the original one.

The fork has 5 additional c files and one header  raspiM* in the host_applications\linux\apps\raspicam folder plus an adjusted CMakeLists.txt to build the raspimjpeg process.

I do occasionally update the main userland files from the raspberry repository but a lot of the changes do not affect the raspimjpeg operation.