silvanmelchior/RPi_Cam_Web_Interface #348: UPDATE ERROR RASPBERRY PI B

perrycannon, 2017-04-29:
When updating I receive an error
./install.sh: line 377: [: ==: unary operator expected
this occurs after reading package lists, building dependency tree, and reading state information.

The Raspberry Pi B was updated this afternoon, and the kernal was updated.

Thank you
Perry

roberttidey, 2017-04-30:
Strange; that part of the install script hasn't changed for a long while. I'll check it out.

KevinOberle, 2017-08-15:
I'm trying to build this for docker using https://github.com/droogmic/rpi-cam-web-docker and am receiving this same error.

Indeed cat /proc/cmdline on my RPi2 is not returning a boardrev.

```
kevin@raspberrypi:~ $ cat /proc/cmdline | awk -v RS=" " -F= '/boardrev/ { print $2 }'
kevin@raspberrypi:~ $ cat /proc/cmdline
bcm2708_fb.fbwidth=656 bcm2708_fb.fbheight=416 bcm2708_fb.fbswap=1 vc_mem.mem_base=0x3dc00000 vc_mem.mem_size=0x3f000000  dwc_otg.lpm_enable=0 console=ttyAMA0,115200 console=tty1 root=PARTUUID=4f85592a-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait
```

roberttidey, 2017-08-15:
Are you using the latest version. I did put a change in so that it ignored a missing boardrev. The line in install.sh should have double [[ which allows a comparison against a blank string to work

KevinOberle, 2017-08-30:
I was not. Now I am and I can confirm this is resolved. Thank you!