silvanmelchior/RPi_Cam_Web_Interface #349: files .h264 are not purged

shinji2009, 2017-05-16:
hi
if .h264 file are stays in media folder after error , it is not purged.
even  got one board with 24gb of .h264 files so theres no place for mp4 files left at all

roberttidey, 2017-05-16:
Normally the .h264 files get deleted after boxing unless one turns boxing off. 

I don't seem to get any left over but I can see it would gradually become a problem if errors let these build up. The reason for not purging just all .h264 files is that some users turn boxing off as they want to box offline.

I will look at a strategy that does .h264 clean up for the normal case.

shinji2009, 2017-05-16:
thank you. and zip files too, please. and maybe just any file type that are not needed in this folder

roberttidey, 2017-05-16:
Couple of purge improvements in the latest version. Note I have to be a bit cautious here as some people store other stuff in the folder from macros etc.

1) .h264 files are now separately checked for when purging or deleting any video related file

2) zip files older than the video purge hours time limit are removed.

shinji2009, 2017-05-24:
thank you. this  improvements will be implemented soon?

roberttidey, 2017-05-24:
They are in the current version

shinji2009, 2017-05-24:
so... h264 file purged only if corresponding mp4 file is present? if theres no mp4 file due to error, h264 file stays forever?

[image]

roberttidey, 2017-05-24:
No. It starts from the thumbnail and deletes all files associated with that including mp4.h264 left over in error.

It won't retrospetively delete old .h264 files left over before the new logic. You have to those manually once. It should clear out from then on.

shinji2009, 2017-06-01:
it looks like these h264 errors appears only when video setting is 1920*1080. i set to 1280*720 a couple days ago and no busy files since then. also a lot of frames are dropped in a row sometimes. should i create a new issue about it?

roberttidey, 2017-06-01:
Yes. Create a new issue, but see if you can highlight anything about your specific set up as I haven't seen any specific problems with the higher resolution.

In particular:

Are you storing to the local SD card or some other external storage?

Is the GPU RAM set to at least 128MB?  ( vcgencmd get_mem gpu )

shinji2009, 2017-06-01:
ok this one should be closed then as i understand.
i'll create a new issue and everything about it will be there.
thank you

closed 2017-06-01 by shinji2009