silvanmelchior/RPi_Cam_Web_Interface #353: How to add overlay image over video stream?

AlexanderShniperson, 2017-06-10:
Hi.

i am trying to add image overlay over live video streaming with using tool `pngview`
that comming from [here](https://github.com/AndrewFromMelbourne/raspidmx/tree/master/pngview)
to run it i use command:
`./pngview -b 0 -d 0 -l 2 -x 0 -y 0 /home/pi/raspberry_logo_small.png`
but nothing happens on the live video

maybe i using wrong layer, can you please take a light on this?

as a note: `pngview` uses Dispmanx to write directly into the GPU of Raspberry

thanks.

roberttidey, 2017-06-11:
I'm not familiar with pngview at the moment. I can see that is has been used to do overlays for other video apps.

raspimjpeg does make heavy use of the GPU and MMAL camera interface to provide multiple streams for preview data, video recording and vector data for motion detection so it may be complex to see if the overlay can be added as well.

Can you give more detail on what you are trying to achieve here as there may be simpler ways to achieve what you want. For example if you just want to overlay something on the preview live view then it may be possible to do that using html5 browser functionality.

AlexanderShniperson, 2017-06-11:
@roberttidey thanks for reply.
I made a [tank robot with](https://pp.userapi.com/c637827/v637827941/5322a/aqBh2p1Ewd4.jpg):
- rpi3
- rpi camera v2
- gy801

i need to have ability to draw some UI directly on the camera layer, [like it does here](https://habrastorage.org/web/d61/efd/78c/d61efd78ca744366b648a12119b108f7.png)

does it possible?

AlexanderShniperson, 2017-06-11:
i forgot to say HTML is not my way, because some data came from two different sides:
1) from the tank robot, like position of tower/chassis
2) from the server about other tanks and their positions
so all of this information need to be synced in one place and this i think must be an image overplay

roberttidey, 2017-06-11:
Actually I think that html overlays would be a good solution here. It would allow the server to be placing the various elements you want (textual and graphic) separately and that would be much more flexible than trying to combine everything into a single image overlay embedded in the preview stream. The server is the single point of synchronisation.

E.g. All Tanks report their positional data to server. Server / Ajax used to real time update the various elements you wish to display. Each tank identity could get its own perspective data if required.

AlexanderShniperson, 2017-06-11:
no, HTML exactly is not my way.
RaspiMJPEG partially have overlay functionality, just start RPI camera web interface and you see there is `annotation` with text `RPi Cam date and time` on the top of given image.

AlexanderShniperson, 2017-06-11:
@roberttidey 
doest it possible to add feature `add/remove image overlay in some x/y position by sending command into FIFO` ?
i am not from C world, so it is very difficult to do it myself

roberttidey, 2017-06-11:
The annotation uses functionality already built into the MMAL camera interface. It is not an overlay. The MMAL software renders a font into the top of each frame and there is no control of positioning. It is possible to add in your own custom values in there using the %a annotation variable. See wiki for details but it is only a text capability, not images.

Adding an image overlay plane is also complicated by the fact that it is already being rendered in the browser, not directly on the desktop window.

AlexanderShniperson, 2017-06-11:
just looked at code, maybe it is possible to postprocess buffer and rewrite some bytes directly in `RaspiMCam.c` after line 71 ?

AlexanderShniperson, 2017-06-12:
i googled a lot yesterday and [found this](https://github.com/waveform80/picamera/issues/199) PiCamera implementation under Python
looks like overlay is possible by MMAL Renderer, but i don't understand what renderer exactly is

roberttidey, 2017-06-12:
Postprocessing the buffer in RaspiCam is not straightforward. The buffer at that point is JPEG compressed so would need to be uncompressed, overlaid and recompressed on the main CPU which would not be a good idea.

The GPU processing chain in raspimjpeg is quite complex as it involves using splitter components, resizers, and encoders. This makes it very efficient but does does mean there is little access to the raw data stream.

It might be possible to introduce an extra rendering element into the processing chain but that is not a trivial exercise given the chain in use already.

On the other hand I did a quick experiment on overlaying elements on the web page. I took the min.php which gives a simple preview only view and did a small modification to overlay an existing jpg. 
[image]
[min1.zip](https://github.com/silvanmelchior/RPi_Cam_Web_Interface/files/1067646/min1.zip)
If you put the min1.php in the camera webfolder then you get the image overlaid onto the live stream.

granpino, 2017-08-15:
I like the idea of overlaying an image into the video. I have been trying to put a cross-hair at the center of the video. I put the sample min1.php file at bar/www/html but do not know what to do next.

roberttidey, 2017-08-15:
If you browse to http://ip/html/min1.php you should see a preview screen with a delete button superimposed.

You can adjust the position by editing the top and left positions in the overlay part of min1.php

You can change the image by putting a different image in the html folder and referring to it instead of delete.jpg

For a cross hair you could use a transparent gif as the image or maybe just draw html lines on top instead of using an overlay image.

The example given was just a demo with a simple preview pane. If you want a cross-hair on top of the normal full web page RPi Cam then the get it right in a min1.php first then edit the same cross hair into the normal index.php page.

archeney, 2017-09-06:
Total newbie here. This is exactly what I am looking for but I cant find the html folder within RPicam or subfolders? I put my min1 variant in the /www folder (where I found min.php), but browsing to http://ip/min1.php gives me a 404, yet http://ip/min.php works just fine. Any help would be appreciated.

granpino, 2017-09-06:
You need to put the min1.php file at /var/www/html folder