silvanmelchior/RPi_Cam_Web_Interface #355: Need a user view that doesn't have controls.

sepeters, 2017-06-12:
Is it possible to have a user view that doesn't have controls.  I'm planning on using this setup for the eclipse in August and want people to be able to open the video on their phones but I do not want them to be able to mess with the video and setting controls.  I would also like that to be wide open and have the admin camera view password protected.

Ideally this would have the user view be the default view for the web server and the admin view be a different/sub URL that is password protected so I can manage that on my tablet for example.

roberttidey, 2017-06-12:
There is a view accessed using ip/min.php which gives just the preview window (no controls) on its own but this does not stop users accessing the other url to get the controls.

I have been working on a version which allows the user display to be set by the login with 3 access levels; preview only, preview plus capture buttons, full control. I have a bit more testing to do but should be able to publish soon.

sepeters, 2017-06-13:
Robert,
Thanks for the prompt response, being a longtime systems engineer what you gave me let me work around the problem...  I left the install in the default of /var/www/html with the apache server root as /var/www...  I password protected it during install so it needs auth to login to the page with settings... then I copied the whole html directory up one level which is NOT THE DEFAULT for recording so those functions wouldn't work anyway but I renamed the index.php and put in a sym link for index.php to point to min.php.  Works like a champ with just the view password free and the default page and the controls password protected.

thanks again, if you get an updated version that has the different privelege levels let me know and I'll be glad to test it out.

Steve

roberttidey, 2017-06-19:
Latest version has support for 3 levels of web functionality. See wiki for details under additions and tricks