silvanmelchior/RPi_Cam_Web_Interface #356: Translate interface

rogercorrea, 2017-06-18:
Please,
I can translate interface for the portuguese language?
I did think make a fork and create a function to globalization the labels.
What you think? It's possible?
Thanks!

roberttidey, 2017-06-19:
Sounds like a good idea, especially to incorporate in main repository rather than make a new fork, so everybody can benefit. You could of course try out changes in a fork and then make a pull request.

It could be a fair bit of work; if you want any help or guidance let me know.

When I was first coding the schedule.php I did start making the labels definable at the top of the program but this didn't get carried through.

Most of the button labels would be fairly straightforward to do. A couple of areas to think about would be
1) Main button action labels (image capture etc) are set dynamically in the javascript according to the current status so that piece would need to go through translation.
2) The schedule config table uses the same names for display in the table as for the keys in the config file itself. It s desirable to keep the keys in the iconfig file unchanged.

It may also be worth thinking how deep you want any translation to be. E.g. it could be just main action buttons or it could include all the settings and config.

ebretteville, 2017-06-19:
I can make the french translation! ;-)

roberttidey, 2017-06-19:
Thanks for that offer. I'd like to see this end up with a translate resource file that would be easy to edit for other languages.

rogercorrea, 2017-07-01:
Thanks! I go make it.

Jacques-Olivier-Farcy, 2020-08-19:
Hi there,

I can also help to the french translation of the interface.
Is there a documentation somewhere to do that ?

roberttidey, 2020-08-19:
Thanks for the offer. There was no progress on the previous part. There is unfortunately no specific documentation on how to do a translate.

My comments on June 19 2017 still apply.

One could start simple and do something just for the main action buttons. As mentioned these are controlled by the javascript script.js file and are in area of code ajax_status.onreadystatechange around line 236. Each line here has a call to a function that includes the text that sets the button label. By changing that so the labels are javascript variables then one could set these at the start to be English, French or other language.

E.g. 

var btn_record_video_start = "record video start";

and then replace each instance of "record video start" by the variable like

setButtonState("video_button", false, btn_record_video_start, "ca 1");

It then becomes simpler to replace the variable definitions by their translations and these can moved out to an external file to make it easier.

If you want to proceed let me know and I'll do the first step of making the script use these variables.

Jacques-Olivier-Farcy, 2020-08-27:
Hi @roberttidey ,

Thank you for your proposal.
And yes, i can do the work for french language when your external file will be ready.

Just let me know