silvanmelchior/RPi_Cam_Web_Interface #358: No apache_getenv in Nginx

konstantinj, 2017-06-23:
Hi,

following error occurs in new version with nginx:
`"PHP message: PHP Fatal error:  Call to undefined function apache_getenv() in /var/www/html/index.php on line 193"`

Currently no time to fix it myself.

roberttidey, 2017-06-23:
Thanks for pointing that out. It is caused by the new support for user the login username to control the features in the user interface. But I only did this for Apache.

A quick workaround is just to replace the line with
$user = "";

I'll see if I can put in an equivalent nginx call

roberttidey, 2017-06-23:
I have changed the code for the moment so that it only tries to get the user name for apache while I try to find equivalent method for nginx

roberttidey, 2017-06-23:
I have put in some code that might work in nginx but I haven't got a nginx install to test with at the moment

konstantinj, 2017-06-23:
Isn't "REMOTE_USER" part of $_SERVER, independent of the used webserver? I just replaced it with this in my fork but haven't tested the user feature.

roberttidey, 2017-06-23:
It doesn't exist on my Apache test which I why I used the other technique.