silvanmelchior/RPi_Cam_Web_Interface #359: Interface not working after Pi Hole install

florisjan78, 2017-06-28:
Hello.

I assumed it was going to be a catastrophe, but installing Pi Hole on my Pi running version 6.2.27 actually worked flawlessly for Pi Hole. The interface however did not appreciate my move but did not give up on me. In fact, everything is still working, except for the user interface. That is the scheduler, motion detection, video recording, the lot, it all works. The preview stream does also work, if I just open it as http://raspberry/cam.jpg. However in the user interface there is no video preview (it stays at loading) and there are no working buttons, except for the schedule settings and download button. Both pages work fine. I have checked all the obvious causes, but like I said everything seems to be running fine in the background. 

Now I have found there have been problems with older versions of Pi Hole with inaccessible websites outside of their /admin dir, but this seems not to be the problem and has afaik already been solved.

I have installed 6.3.13 over the old version after installing Pi Hole, but the situation is still the same.  I was not really expecting either of the programs to work at all so I am quite satisfied, however solving this would be perfect.

roberttidey, 2017-06-29:
First try accessing http://raspberry/min.php This gives just basic preview screen with no controls.

If that doesn't work then I think one has to systematically work one's way up from basics to see where the web interface is broken. One can try putting in a simple html page in the web folder and making sure that works. Next step would be a basic test php page. E.g. `<?php
echo "it works";
?>`

It is also good to turn on development tools in the browser so that one can see what is going on.

florisjan78, 2017-06-30:
Thanks for pointing me in right direction, I overlooked the blocking of .js files.

Solution: In lighttpd.conf the following line must be commented out so .js are no longer blocked.

`# Rewite js requests, must be out of $HTTP block due to bug #2526`
`#url.rewrite = ( "^(?!/admin/).*\.js$"  => "pihole/index.js"   )`

closed 2017-06-30 by florisjan78