silvanmelchior/RPi_Cam_Web_Interface #360: Conflict both Raspimjpeg and raspistill

rogercorrea, 2017-07-01:
Please,

I created one solution using Rpi Cam and raspistill for capture and video monitoring.
But I need of image capture using raspistill and I only can if stop the raspimjpeg. If I don't stop the raspistill returned a error:

mmal: mmal_vc_component_enable: failed to enable component: ENOSPC
mmal: camera component couldn't be enabled
mmal: main: Failed to create camera component
mmal: Failed to run camera app. Please check for firmware updates

I disabled the raspimjpeg before use the raspistill using sudo pkill raspimjpeg and created a separated script for call raspimpjeg again (the script is the same content that rc.local).

It's right?
Thanks!

rogercorrea, 2017-07-01:
Script that I used:

`sudo pkill raspimjpeg
sudo raspistill -o test.jpg
su -c 'raspimjpeg > /dev/null 2 &
exit 0`

roberttidey, 2017-07-01:
Only one software can connect to the camera MMAL layer at any one time. 

If you have the RPI Web interface running then that will normally stop raspivid and raspistill being used as you have discovered.

You can temporarily disconnect the raspimjpeg from the camera by pressing the stop button on the web interface. THis doesn't kill the raspimjpeg process but just tells it to disconnect so that other software can use the camera. You can then press the start button to resume operation.

You can achieve the same effect from a command line or script by sending in commands to the command pipe e.g `echo 'ru 0' >/var/www/html/FIFO` stops the camera interface and `echo 'ru 1' >/var/www/html/FIFO` starts it up. These assume install is under html; if not then adjust accordingly.

Note also that you can use the im command to grab a still in exactly the same way as raspistill without having to stop anything.

rogercorrea, 2017-07-01:
My solution is automatic. Not have user to press button.
For it I created the script that stop the raspimjpeg process and execute the raspistill.
I understand your answer and thanks for it. 

So, is good that all know about it.
Thanks!

closed 2017-07-01 by rogercorrea

roberttidey, 2017-07-02:
Reason for maybe preferring to do it via the command pipe in an automatic context is that it allows raspimjpeg to stop any current operations and close the interface cleanly instead of killing the process dead no matter what raspimjpeg is doing at the time.