silvanmelchior/RPi_Cam_Web_Interface #364: Motion detection: never stops recording after triggered

playahata, 2017-07-14:
The stream I see looks good when it's not recording, but after motion is detected (during recording) it  stream changes and the foreground light starts to flicker on the screen. I think this causes the motion detection to keep detecting motion even though there is non. why does the stream change when it is recording, i.e. why does it look different (worse) on the screen than when not recording?

roberttidey, 2017-07-14:
Stream shouldn't change significantly under reasonable lighting conditions but may go darker when video is recording under poor lighting conditions. This is because the normal 25 fps frame rate of video stops the camera making some of the lighting adjustments.

If you are seeing preview change under reasonable lighting then please let me have any settings changed you have applied so I can check those out. These are found in the uConfig file in the web folder (e.g. /var/www/html

playahata, 2017-07-18:
turns out the issue was "shutter speed" parameter. when I set it to the max of 330000, the flickering dissapeared when recording and no rogue motion is detected. so everything works just fine. one more thing, how can I turn off the generation of thumbnails for each video? I followed the instructions in raspimjpeg file, but this does not work. thx.

roberttidey, 2017-07-18:
The thumbnails are a key item of the interface. They are used to index and manage all the files (videos, stills and timelapses) even when stored in nested subfolders. 

Why do you want to turn them off? 

The thumb_gen parameter in raspimjpeg does allow them to be selectively turned off for video, images or thumbnails. This would normally be done only if you want to manage the files yourself and not use the web interface.

Note you have to edit the /etc/raspimjpeg file AND the camera software needs to be restarted (use ./stop.sh and .start.sh in the install directory

playahata, 2017-07-19:
thumb_gen setting worked, I am indeed managing the files myself. In this case, the thumbnails are just something one has to delete every time, so I'd rather not generate them in the first place. thanks!