silvanmelchior/RPi_Cam_Web_Interface #365: Only see part of image in preview with camera module V2

tom-2015, 2017-07-17:
I'm using the new V2 camera module but in the preview I can only see the left upper part of the total image the camera is capturing. When I take a picture it does record the full image.

It's like the preview picture is only capturing the old 5MP resolution, how can I adjust this?

roberttidey, 2017-07-17:
That is strange as others use v2 OK.

The preview image is derived from the video format.

First try browsing to http://cameraip/folder/cam_pic.php where you need to replace the cameraip and folder if used. This should give you one snapshot of the images used to produce the preview stream. Does that look OK or is it cut off as well?

Try also changing the video format under camera settings.

If you have done any custom settings then please upload content of uConfig file in the web folder so I can check that out.

tom-2015, 2017-07-17:
Changing the video resolution solved the problem but FPS is really low now, should have taken a v1 camera :(

roberttidey, 2017-07-17:
That still doesn't sound right. V2 should give same performance as V1.

Do you mean fps on the preview or fps on the actual video recording?

The preview fps should be the video recording fps divider by the preview 'divider' setting. This defaults to 1 so a 25fps video rate should give a 25 fps preview. Bandwidth can become an issue if using a wireless connection. You can check the fps on preview by turning on developer tools in the browser and leaving the preview at the default setting as each preview frame is then fetched separately and you can see those requests being handled on the network tab.

tom-2015, 2017-07-18:
No it was the preview window but today it is working fine, maybe it was because of the F12 debug/network window in Firefox was open yesterday. I'll see if it happens again coming days.

Thanks for the help anyway.

mfr69bs, 2017-12-27:
Same issue here on a Raspberry Pi 3, Camera V1.3, running on Raspbian Stretch:
Only the upper left corner of the stream is visible. Also on a recording and on browsing to cam_pic.php.

Changing the resolution does not fix it.

width 512
quality 10
divider 1

video_width 1920
video_height 1080
video_fps 25
video_bitrate 17000000
video_buffer 0

h264_buffer_size 131072
h264_buffers 0
video_split 0

MP4Box background
MP4Box_fps 25
MP4Box_cmd (set -e;MP4Box -fps %i -add %s %s > /dev/null 2>&1;rm "%s";) &

image_width 2592
image_height 1944
image_quality 10

roberttidey, 2017-12-28:
Check camera connections are secure at both the Pi and camera ends and press down on the flat end of the small ribbon cable on the camera itself.

Check whether raspivid captures the whole frame normally.

tom-2015, 2017-12-28:
You need to change the resolution in the RPi Cam Control web interface not in /boot/config.txt
Click the camera settings button and enter video / picture resolution 3280x2464

mfr69bs, 2017-12-28:
@roberttidey 

> Check camera connections are secure at both the Pi and camera ends and press down on the flat end of the small ribbon cable on the camera itself.

That was the trick. THX!

@tom-2015 

I did not write anything about /boot/config.txt. So please do not just write something about it.

There is also something below the "surface" (RPi Cam Control web interface) for raspimjpeg. So called configuration files ;-)

tmo26, 2018-10-27:
RPi Cam Control v6.4.31
Cam module v2

Video Res = 1920x1080
Image Res = 1920x1080

**Expected output:** Preview shows the same FOV as the captured image

**Observed output:** Preview shows significantly smaller FOV as the captured image

[image]

**Solution:** Set Video Res to 1640x922 -> preview shows the same viewing angles (horizontal + vertical) as the captured image

roberttidey, 2018-10-27:
Yes. This is due to the way the GPU handles the FOV.

The preview is based on the video resolution. The 1920 x 1080 selects 1920 x 1080 pixels from the camera to process. The still image capture takes a larger region and then scales it down to fit the 1920 x 1080 frame

andyryan, 2020-04-28:
Apologies Sorry to resurrect an old issue, but I've been playing with the software today to use as the stream for 3d printer recording, and the issue with the above comment is that 720p (16:9) does not have the same issue as the 1080p. While 1080p corps the image as show above, 720p does not, so it is not due to the aspect ratio, or number of pixels selected (you would expect 720p to select an even smaller FOV based on that argument)

The way I got around this was by setting the video capture to 1440x1080 only then do i get something that can be cropped to 1080p

roberttidey, 2020-04-28:
When I select the HD Ready 720p preset (1280 x 720) then I get exactly the same FOV preview as when I select Full HD (1920 x 1080). This is what I would expect as the aspect ratio is selecting the portion of the camera sensor that can be used to give that aspect ratio. The sensor is intrinsically 4:3 so only video formats with that aspect ratio will show the full sensor FOV.

andyryan, 2020-04-28:
[image]
[image]
[image]

Above are three images that I hope better illustrate the issue. The top most is 1440x1080 (a 4:3 ratio). As expected you get the fill field of view of the camera

The second is 720p (16:9) As expected, you get the full horizontal FOV, but the top and bottom portion of the images are cropped out.

The third is 1080p. You would expect the same FOV as 720p, with a higher pixel count. However the FOV is also reduced in the horizontal direction. 

Note: this is only happening for video. It does not occur when setting the image resolution to 1920x1080.

roberttidey, 2020-04-28:
Under further investigation

roberttidey, 2020-04-28:
OK. Check out https://picamera.readthedocs.io/en/release-1.12/fov.html  for an explanation of the FOV for different modes

andyryan, 2020-04-29:
Colour me corrected. That is the strangest way of a camera handing resolutions I've ever seen.

roberttidey, 2020-04-29:
It is all to do with only dealing with integer pixels and not interpolating which would be very processor intensive. So with 1280 width it is possible to bin (add two pixels together) and get the nearly the whole FOV. With 1920 x 1080p that is not possible so they revert using single pixels and cropping from the centre.

As you can see the effects are different between V1 and V2 camera which have different pixel counts.

So for 16:9 capture with V1 then HD Ready gives nearly the best FOV. For V2 you can go to 1640 x 922. 

If you are using v2 then you can use the uPresets.html to load in a different sets of preset resolutions for convenience. Look at the example uPresetsP-IMX219.html. If the option data in this is put into a uPrests.html file then this will load and set new presets.