silvanmelchior/RPi_Cam_Web_Interface #366: failure to install

ongdextersg, 2017-07-19:
E: Failed to fetch http://mirrordirector.raspbian.org/raspbian/pool/main/d/dialog/dialog_1.2-20140911-1_armhf.deb  Hash Sum mismatch

E: Unable to fetch   some archives, maybe run apt-get update or try with --fix-missing?

Hi - I got th error message when I run ./install.sh

How can I fix that?  I am new to Raspberry.  Thanks.

roberttidey, 2017-07-19:
That error is when it is trying to install the standard dialog package which the installer uses to display menus during the install process.

It could be caused by one of 2 things. 

First you must do the
sudo apt-get update
sudo apt-get dist-upgrade

as described in the wiki. Otherwise it is possible that the package you are trying to install is not in synce with your OS.

Second, sometimes there can be a temporary failure to access the package archives online. Just try it again.

ongdextersg, 2017-07-19:
Bob - Thank you!  It worked like a charm!  Many thanks.

-Dexter

closed 2017-07-19 by ongdextersg