silvanmelchior/RPi_Cam_Web_Interface #370: Can this frame take 10fps and I want  see you raspimjpeg.c ?

dtwangquan, 2017-09-12:
I have two question:

1、I want use this frame's timelapse to take picture 10fps, but it only can 2fps.

2、I want change the preview resolution, but the highest limit is 1024 (Width (128...1024)),I want cant find the pv command from RaspiMJPEG.c https://github.com/silvanmelchior/userland/blob/master/host_applications/linux/apps/raspicam/RaspiMJPEG.c
so, I think you have owne RaspiMJPEG.c ， I want it

my english is not very well, hope you can understand，Expect your reply，  thank you！

anomaly256, 2017-09-12:
Hi, do you mean this?  https://github.com/roberttidey/userland/blob/master/host_applications/linux/apps/raspicam/RaspiMJPEG.c

roberttidey, 2017-09-13:
Time lapse uses the still image facility and it is difficult to take these at a high rate. I would suggest using the video facility set to 10 fps.

The preview width is limited to the video width.

The source for the raspimjeg executable is given in a link in the wiki. It is NOT the silvanmelchior github; that is a very old original one.
https://github.com/roberttidey/userland/tree/master/host_applications/linux/apps/raspicam

The files of interest start with RaspiM