silvanmelchior/RPi_Cam_Web_Interface #371: Fails on raspbian stretch

Lewiscowles1986, 2017-09-21:
raspbian stretch defaults to php7 so this fails. The best way I got it to work was to install, allow it to fail, then install nginx and php7, manually edit the default to enable php7 by commenting out the lines and adding index.php as a root.

The camera module I got apparently has some issues so I won't know if it's 100% before I get a new sensor, but the raspbian errors are real, so I thought I'd report them here.

It's not really PR ready as I've no clue which lines failed other than apt-get install lines, or what the setup does besides destroying nginx config (could `rm /etc/nginx/sites-enabled/default` instead of nuking it's contents).

roberttidey, 2017-09-21:
Thanks. I am currently doing some tests with latest stretch raspbian to try to resolve the issues with this change.

mfraser, 2017-09-21:
I installed 6.3.15 on a fresh install of Stretch lite at the weekend and didn't experience any problems with PHP. dpkg -l php* shows:
php5            5.6.22+dfsg-2
php5-cli        5.6.22+dfsg-2+b3
php5-common     5.6.22+dfsg-2+b3
php5-json       1.3.9-1
php5-readline   5.6.22+dfsg-2+b3

roberttidey, 2017-09-21:
Thanks for that report. Can you tell me which web server you chose (apache, nginx lighttpd)?

I am also experimenting changing the php to 7.0

Lewiscowles1986, 2017-09-21:
@mfraser I was on stretch full, but I'm happy to retry with lite (shouldn't be any difference)

mfraser, 2017-09-22:
@roberttidey I am using apache.

ebretteville, 2017-09-22:
I'm on stretch too with apache2 and it's working well.

Lewiscowles1986, 2017-09-22:
@ebretteville & @mfraser are you both running the install script? What manual intervention are you performing?

ebretteville, 2017-09-22:
Fresh install, update && upgrade, git pull and the script, thats it.

Lewiscowles1986, 2017-09-22:
Seen it https://github.com/silvanmelchior/RPi_Cam_Web_Interface/commit/681bda993407c87ccf84f6ddc13e34da89c9ee2b. I'll try again and close if it fixes.

roberttidey, 2017-09-22:
Also note that the latest install script has option to use php7 or php 5. I have only tested php7 with apache installs so far.

ebretteville, 2017-09-22:
Will try to update next week from 6.3.14 to 6.14 and try to use PHP7 that require less ressources. We'll keep you posted

Lewiscowles1986, 2017-09-22:
what I'm doing is baking an image before running install, so I can restore that, git pull, and retry. I'll run all the web servers under php7.

The initial idea was to get steps to setup without web-server etc, so it could be barebones dropped into an existing pi without playing around with web-server packages or config.

Lewiscowles1986, 2017-09-22:
So I'm running this under stretch-lite as @mfraser mentioned that was working. I think it might be an idea to add `--no-install-recommends` to the `apt-get install` to keep it light (I'm happy to PR any suggestions btw)

```
The following additional packages will be installed:
  apache2-bin apache2-data apache2-utils ffmpeg fontconfig-config
  fonts-dejavu-core gpac-modules-base liba52-0.7.4 libaacs0 libapr1
  libaprutil1 libaprutil1-dbd-sqlite3 libaprutil1-ldap libass5 libasyncns0
  libavc1394-0 libavcodec57 libavdevice57 libavfilter6 libavformat57
  libavresample3 libavutil55 libbdplus0 libbluray1 libbs2b0 libcaca0 libcairo2
  libcdio-cdda1 libcdio-paranoia1 libcdio13 libchromaprint1 libdc1394-22
  libdrm-amdgpu1 libdrm-freedreno1 libdrm-nouveau2 libdrm-radeon1 libebur128-1
  libegl1-mesa libfaad2 libfftw3-double3 libflac8 libflite1 libfontconfig1
  libfreenect0.5 libfribidi0 libgbm1 libgl1-mesa-dri libgl1-mesa-glx
  libglapi-mesa libglu1-mesa libgme0 libgpac4 libgraphite2-3 libgsm1
  libgstreamer1.0-0 libharfbuzz0b libice6 libiec61883-0 libjack-jackd2-0
  libllvm3.9 liblua5.2-0 libmad0 libmariadbclient18 libmp3lame0 libmpg123-0
  libogg0 libopenal-data libopenal1 libopencv-core2.4v5 libopencv-imgproc2.4v5
  libopenjp2-7 libopenmpt0 libopus0 libpgm-5.2-0 libpixman-1-0 libpostproc54
  libpq5 libpulse0 libraw1394-11 librubberband2 libsdl1.2debian libsdl2-2.0-0
  libsensors4 libshine3 libsm6 libsnappy1v5 libsndfile1 libsndio6.1
  libsodium18 libsoxr0 libspeex1 libssh-gcrypt-4 libswresample2 libswscale4
  libtheora0 libtwolame0 libtxc-dxtn-s2tc libva-drm1 libva-x11-1 libva1
  libvdpau-va-gl1 libvdpau1 libvorbis0a libvorbisenc2 libvorbisfile3 libvpx4
  libwavpack1 libwayland-client0 libwayland-cursor0 libwayland-egl1-mesa
  libwayland-server0 libwebp6 libwebpmux2 libx11-xcb1 libx264-148 libx265-95
  libxcb-dri2-0 libxcb-dri3-0 libxcb-glx0 libxcb-present0 libxcb-render0
  libxcb-shape0 libxcb-shm0 libxcb-sync1 libxcb-xfixes0 libxcursor1
  libxdamage1 libxfixes3 libxi6 libxinerama1 libxkbcommon0 libxrandr2
  libxrender1 libxshmfence1 libxss1 libxtst6 libxv1 libxvidcore4 libxxf86vm1
  libzmq5 libzvbi-common libzvbi0 mesa-va-drivers mesa-vdpau-drivers
  mysql-common php-common php7.0-common php7.0-json php7.0-opcache
  php7.0-readline ssl-cert va-driver-all vdpau-driver-all x11-common
Suggested packages:
  www-browser apache2-doc apache2-suexec-pristine | apache2-suexec-custom
  ffmpeg-doc gstreamer1.0-plugins-base php-pear libbluray-bdj libfftw3-bin
  libfftw3-dev alsa-base jackd2 libportaudio2 opus-tools pulseaudio
  libraw1394-doc lm-sensors sndiod speex default-mysql-client
  postgresql-client openssl-blacklist
The following NEW packages will be installed:
  apache2 apache2-bin apache2-data apache2-utils ffmpeg fontconfig-config
  fonts-dejavu-core gpac gpac-modules-base gstreamer1.0-tools liba52-0.7.4
  libaacs0 libapache2-mod-php7.0 libapr1 libaprutil1 libaprutil1-dbd-sqlite3
  libaprutil1-ldap libass5 libasyncns0 libav-tools libavc1394-0 libavcodec57
  libavdevice57 libavfilter6 libavformat57 libavresample3 libavutil55
  libbdplus0 libbluray1 libbs2b0 libcaca0 libcairo2 libcdio-cdda1
  libcdio-paranoia1 libcdio13 libchromaprint1 libdc1394-22 libdrm-amdgpu1
  libdrm-freedreno1 libdrm-nouveau2 libdrm-radeon1 libebur128-1 libegl1-mesa
  libfaad2 libfftw3-double3 libflac8 libflite1 libfontconfig1 libfreenect0.5
  libfribidi0 libgbm1 libgl1-mesa-dri libgl1-mesa-glx libglapi-mesa
  libglu1-mesa libgme0 libgpac4 libgraphite2-3 libgsm1 libgstreamer1.0-0
  libharfbuzz0b libice6 libiec61883-0 libjack-jackd2-0 libllvm3.9 liblua5.2-0
  libmad0 libmariadbclient18 libmp3lame0 libmpg123-0 libogg0 libopenal-data
  libopenal1 libopencv-core2.4v5 libopencv-imgproc2.4v5 libopenjp2-7
  libopenmpt0 libopus0 libpgm-5.2-0 libpixman-1-0 libpostproc54 libpq5
  libpulse0 libraw1394-11 librubberband2 libsdl1.2debian libsdl2-2.0-0
  libsensors4 libshine3 libsm6 libsnappy1v5 libsndfile1 libsndio6.1
  libsodium18 libsoxr0 libspeex1 libssh-gcrypt-4 libswresample2 libswscale4
  libtheora0 libtwolame0 libtxc-dxtn-s2tc libva-drm1 libva-x11-1 libva1
  libvdpau-va-gl1 libvdpau1 libvorbis0a libvorbisenc2 libvorbisfile3 libvpx4
  libwavpack1 libwayland-client0 libwayland-cursor0 libwayland-egl1-mesa
  libwayland-server0 libwebp6 libwebpmux2 libx11-xcb1 libx264-148 libx265-95
  libxcb-dri2-0 libxcb-dri3-0 libxcb-glx0 libxcb-present0 libxcb-render0
  libxcb-shape0 libxcb-shm0 libxcb-sync1 libxcb-xfixes0 libxcursor1
  libxdamage1 libxfixes3 libxi6 libxinerama1 libxkbcommon0 libxrandr2
  libxrender1 libxshmfence1 libxss1 libxtst6 libxv1 libxvidcore4 libxxf86vm1
  libzmq5 libzvbi-common libzvbi0 mesa-va-drivers mesa-vdpau-drivers motion
  mysql-common php-common php7.0 php7.0-cli php7.0-common php7.0-json
  php7.0-opcache php7.0-readline ssl-cert va-driver-all vdpau-driver-all
  x11-common zip

```

Lewiscowles1986, 2017-09-22:
Apache setup with defaults seems to create files in folder outside of web-root 

[image]

roberttidey, 2017-09-22:
As installed with script the folder is /var/www/myfolder where myfolder is defined in script set up (default html)

Lewiscowles1986, 2017-09-22:
@roberttidey what I'm showing is that without an existing apache, with no change, other than to set php to 7 (as it was blank), the config for apache seems to put files into `/var/www/html` contents, but then relocate web root to `/var/www`, not `/var/www/html`

roberttidey, 2017-09-22:
That is correct. The web root is deliberately left at /var/www. To access the camera one uses http://ip/html or whatever other folder one wishes to use.

The reasoning here is that some people also use the apache server for other accesses and want to keep those in different folders. One could use different port addresses to distinguish between these and at one point a couple of years ago it was like that but most people didn't want to put into different port addresses particularly if they were using port forwarding, SO it was changed to keep the web root constant.

For those, including myself, who use this as a single app device then one sets the folder to empty so the files go into /var/www and the access is http://ip

Lewiscowles1986, 2017-09-22:
So it's meant to be that way for apache only? (I'm just trying to understand, raise issues related to stretch here as an epic, so nothing is missed, all is documented somewhere)

Lewiscowles1986, 2017-09-22:
nginx default install with php7 seems to follow apache

default 403 for `/`
[image]

502 for `/html` (likely socket-file issue from virtualhost)
[image]

Lewiscowles1986, 2017-09-22:
looks like https://github.com/silvanmelchior/RPi_Cam_Web_Interface/blob/master/etc/nginx/sites-available/rpicam.1#L16 is the culprit, adjusting fixes

roberttidey, 2017-09-22:
Thanks. I have done that merge.

Lewiscowles1986, 2017-09-22:
I think if the /html/ is intentional, I'm happy to close this :smile:

roberttidey, 2017-09-22:
OK.

closed 2017-09-22 by Lewiscowles1986