silvanmelchior/RPi_Cam_Web_Interface #375: nginx workaround

ZehelSitchel, 2017-10-12:
First, thank you to the developers who wrote this incredible script. Last year when I originally used it, it worked flawlessly and I would have never had any issues if a power outage hadn't corrupted my SD card.

Unfortunately I don't have the know how to edit the script and help but here are the steps I took to get RPi_Cam_Web_Interface to work for me today.

First I started with a fresh install of raspbian (2017-09-07-raspbian-stretch). On first boot I only enabled the VNC and Camera Interfaces. Then I ran the following commands to remove unwanted programs, clear up some space, update, and install nginx.

```
$ sudo apt-get --purge remove wolfram-engine minecraft-pi* libreoffice* scratch* sonic-pi dillo gpicview $ oracle-java8-jdk openjdk-7-jre oracle-java7-jdk openjdk-8-jre
$ sudo apt-get clean
$ sudo apt-get autoremove
$ sudo apt-get update
$ sudo apt-get upgrade
$ sudo apt-get dist-upgrade
$ sudo rpi-update
$ sudo apt-get install php-fpm nginx apache2-utils
```

nginx is now installed and you can access the Welcome Page on the raspberry pi in your web browser by either by entering **localhost** or **127.0.0.1** in the navigation bar.

```
$ cd /etc/nginx/sites-enabled
$ sudo nano default
```

Find the line that looks like: **index index.html index.htm index.nginx-debian.html;**
Make it look like: 

`index index.php index.html index.htm;`

Scroll down and find: **#location ~ \.php$ {** 
Make it look like this:
```
location ~ \.php$ {
		include snippets/fastcgi-php.conf;
	#
	#	# With php-fpm (or other unix sockets):
		fastcgi_pass unix:/var/run/php/php7.0-fpm.sock;
	#	# With php-cgi (or other tcp sockets):
	#	fastcgi_pass 172.0.0.1:9000;
	}
```

Push "CTRL + O"  to save and "CTRL + X" to exit.

We are now ready to download RPi_Cam_Web_Interface. I chose to download to Downloads in my home directory
```
$ cd /home/pi/Downloads
$ git clone https://github.com/silvanmelchior/RPi_Cam_Web_Interface.git
$ sudo apt-get install gpac motion zip libav-tools gstreamer1.0-tools
$ cd RPi_Cam_Web_Interface/www
$ sudo cp -r * /var/www/html/
$ sudo systemctl reload nginx
```

If you reload your web page you should see the RPi_Cam_Web_Interface site but the camera stream won't load. To fix this run the installer.

`$ ./install.sh`

These are the options I used and my installation works. If you tell the installer you are using nginx, something breaks and you'll have to start over. (I had to restart with a new image of the OS otherwise I spent way to long trying to figure out why nginx no longer worked....)

```
Cam subfolder:				html
Autostart: (yes/no)			yes
Server: (apache/nginx/lighttpd)		
Webport:				80
User: (blank=nologin)				
jpglink:(yes/no)			no
phpversion: (5/7)			7
```

If you want authentication for your web site follow the steps below. We installed apache2-tools at the beginning so that we can use htpasswd. You will be prompted to enter a password after the command.

```
$ sudo htpasswd -c /etc/nginx/.htpasswd "USER_NAME_OF_YOUR_CHOICE"
Enter a password:
Re-enter password:
$ cd /etc/nginx/sites-enabled
$ sudo nano default
```

Find **server_name _;**
Under it add:

```
auth_basic "Restricted";
auth_basic_user_file /etc/nginx/.htpasswd;
```

Finally reload nginx.
`$ sudo systemctl reload nginx`

roberttidey, 2017-10-12:
Thanks for the detailed list of how you got it to work.

Normally it should work just by running ./install.sh without pre-installing any other prerequisites. Installing stuff like the web server first can confuse the install script as I think you found by putting nginx on first.

ZehelSitchel, 2017-10-12:
@roberttidey, I tried many different variations trying to get it to work. When I simply ran the script right after a clean raspbian install,  I would try to access the site and get **403 Forbidden** error. 

After checking the config in /etc/nginx/sites-available, I noticed that the root was /var/www. The default root in the script is /var/www/html. I changed the root to /var/www/html and was prompted for my username and password, previously it didn't ask for it. After entering them I get a **500 Internal Server Error**. 

Trying to troubleshoot the 2 errors for 2 of my evenings is what let me to figuring out the workaround.

roberttidey, 2017-10-12:
The install always leaves the server root as /var/www. The web code is placed under whatever folder is specified in the install.

So if this is html then the client should browse to http://ip/html
Likewise if folder is set to blank then the code goes directly into /var/www and the browse is to http://ip

The reason for doing it like this is for when people want other web apps in the server under different folders but all using the same port address (e.g. 80)

ZehelSitchel, 2017-10-12:
I just did a clean install. It seems to be an issue with authentication.

Doing a clean install without authentication does work. Though I still had to go and edit the config to change root from `root /var/www` -> `root /var/www/html `

ZehelSitchel, 2017-10-12:
Just saw your update. Going to **localhost/html** or **http://ip/html** does allow me access... Maybe I didn't remember that part from my first install a year ago. : /

However I still get a **500 Internal Server Error** when using authentication.
Running my own `sudo htpasswd -c /usr/local/.htpasswd "USERNAME"` resolves that.

roberttidey, 2017-10-12:
Yes. Authentication in the install was primarily set up for Apache and may have issues with latest versions of nginx which have changed around a bit. 

Currently the script uses
sudo htpasswd -b -B -c /usr/local/.htpasswd $user $webpasswd

I'll try to check that out further.

closed 2017-10-12 by ZehelSitchel