silvanmelchior/RPi_Cam_Web_Interface #376: First time install works, on reboot no more image stream

hazardlabs, 2017-10-19:
When installing RPicam using the 'sudo ./install.sh' command,RPi Cam installs and works as expected.

However as soon as I restart my Pi Zero W, I will no longer be able to view images, however the web server is still running.

Apache2 error log as follows:

`[Thu Oct 19 13:22:46.592192 2017] [:error] [pid 384] [client xxx.xx.xxx.xx:16395] PHP Warning:  file_get_contents(status_mjpeg.txt): failed to open stream: No such file or directory in /var/www/status_mjpeg.php on line 6, referer: www.example.net`

*I've removed my ip and domain name from the error log

If I re-run the installer, keeping all settings the same I can get the images back, until I then need to reboot the system again.

The only difference when rerunning the installer is this message in the terminal:

`cp: not writing through dangling symlink '/var/www/status_mjpeg.txt'`

Any ideas to what may be causing this behavior?

hazardlabs, 2017-10-19:
I can get the images to reappear if I manually run the ./start.sh script in my home folder

roberttidey, 2017-10-19:
Are you sure you have autostart set to yes during the install. It sounds like the raspimjpeg process is not starting on boot.

This would mean no status_mjpeg.txt is being created and explains the symlink error.

After a boot with autostart yes then if you do a ps -A command you should see 2 raspimjpeg processes in the list

hazardlabs, 2017-10-20:
I can confirm that Autostart was set to yes, both on the original install and subsequent installs.

On boot up I have the following output after a ps -A command:

  PID TTY          TIME CMD
    1 ?        00:00:03 systemd
    2 ?        00:00:00 kthreadd
    3 ?        00:00:00 ksoftirqd/0
    4 ?        00:00:00 kworker/0:0
    5 ?        00:00:00 kworker/0:0H
    6 ?        00:00:00 kworker/u2:0
    7 ?        00:00:00 lru-add-drain
    8 ?        00:00:00 kdevtmpfs
    9 ?        00:00:00 netns
   10 ?        00:00:00 khungtaskd
   11 ?        00:00:00 oom_reaper
   12 ?        00:00:00 writeback
   13 ?        00:00:00 kcompactd0
   14 ?        00:00:00 crypto
   15 ?        00:00:00 bioset
   16 ?        00:00:00 kblockd
   17 ?        00:00:00 watchdogd
   18 ?        00:00:00 kworker/0:1
   19 ?        00:00:00 rpciod
   20 ?        00:00:00 xprtiod
   21 ?        00:00:00 kswapd0
   22 ?        00:00:00 nfsiod
   32 ?        00:00:00 kthrotld
   33 ?        00:00:00 bioset
   34 ?        00:00:00 bioset
   35 ?        00:00:00 bioset
   36 ?        00:00:00 bioset
   37 ?        00:00:00 bioset
   38 ?        00:00:00 bioset
   39 ?        00:00:00 bioset
   40 ?        00:00:00 bioset
   41 ?        00:00:00 bioset
   42 ?        00:00:00 bioset
   43 ?        00:00:00 bioset
   44 ?        00:00:00 bioset
   45 ?        00:00:00 bioset
   46 ?        00:00:00 bioset
   47 ?        00:00:00 bioset
   48 ?        00:00:00 bioset
   49 ?        00:00:00 bioset
   50 ?        00:00:00 bioset
   51 ?        00:00:00 bioset
   52 ?        00:00:00 bioset
   53 ?        00:00:00 bioset
   54 ?        00:00:00 bioset
   55 ?        00:00:00 bioset
   56 ?        00:00:00 bioset
   57 ?        00:00:00 iscsi_eh
   58 ?        00:00:00 dwc_otg
   59 ?        00:00:00 DWC Notificatio
   60 ?        00:00:00 VCHIQ-0
   61 ?        00:00:00 VCHIQr-0
   62 ?        00:00:00 VCHIQs-0
   63 ?        00:00:00 VCHIQka-0
   64 ?        00:00:00 SMIO
   65 ?        00:00:00 kworker/0:2
   66 ?        00:00:00 irq/86-mmc1
   67 ?        00:00:00 kworker/0:3
   68 ?        00:00:00 bioset
   69 ?        00:00:00 mmcqd/0
   70 ?        00:00:00 ipv6_addrconf
   82 ?        00:00:00 kworker/u2:1
   88 ?        00:00:00 systemd-journal
  107 ?        00:00:00 f2fs_gc-179:2
  111 ?        00:00:00 f2fs_flush-179:
  125 ?        00:00:00 systemd-udevd
  144 ?        00:00:00 kworker/0:1H
  160 ?        00:00:00 cfg80211
  162 ?        00:00:00 brcmf_wq/mmc1:0
  163 ?        00:00:00 brcmf_wdog/mmc1
  182 ?        00:00:00 kworker/0:4
  191 ?        00:00:00 systemd-timesyn
  212 ?        00:00:00 thd
  214 ?        00:00:00 cron
  226 ?        00:00:00 rsyslogd
  231 ?        00:00:00 systemd-logind
  235 ?        00:00:00 dbus-daemon
  241 ?        00:00:00 rngd
  295 ?        00:00:00 sshd
  340 tty1     00:00:00 agetty
  372 ?        00:00:00 wpa_supplicant
  374 ?        00:00:00 apache2
  375 ?        00:00:00 kworker/u2:2
  389 ?        00:00:00 apache2
  391 ?        00:00:00 apache2
  392 ?        00:00:00 apache2
  395 ?        00:00:00 apache2
  397 ?        00:00:00 apache2
  406 ?        00:00:00 dhclient
  426 ?        00:00:00 apache2
  427 ?        00:00:00 apache2
  474 ?        00:00:00 sshd
  479 ?        00:00:00 systemd
  482 ?        00:00:00 (sd-pam)
  489 ?        00:00:00 sshd
  492 pts/0    00:00:00 bash
  501 pts/0    00:00:00 ps

As can be seen, no raspimjpeg in the list. 

What is modified on the system to enable the autostart? I want to confirm that the settings are present, and if not manually add them and see the result.

Thank you for your help thus far.

roberttidey, 2017-10-20:
When autostart is set to yes then the installer adds start up code /etc/rc.local which should be executed during boot.

This file should contain a section like
 `#START RASPIMJPEG SECTION
mkdir -p /dev/shm/mjpeg
chown www-data:www-data /dev/shm/mjpeg
chmod 777 /dev/shm/mjpeg
sleep 4;su -c 'raspimjpeg > /dev/null 2>&1 &' www-data
if [ -e /etc/debian_version ]; then
  sleep 4;su -c 'php /var/www/schedule.php > /dev/null 2>&1 &' www-data
else
  sleep 4;su -s '/bin/bash' -c 'php /var/www/schedule.php > /dev/null 2>&1 &' w$
fi
#END RASPIMJPEG SECTION
`

hazardlabs, 2017-10-20:
Doing a nano /etc/rc.local gives:

#START RASPIMJPEG SECTION
mkdir -p /dev/shm/mjpeg
chown www-data:www-data /dev/shm/mjpeg
chmod 777 /dev/shm/mjpeg
sleep 4;su -c 'raspimjpeg > /dev/null 2>&1 &' www-data
if [ -e /etc/debian_version ]; then
  sleep 4;su -c 'php /var/www/schedule.php > /dev/null 2>&1 &' www-data
else
  sleep 4;su -s '/bin/bash' -c 'php /var/www/schedule.php > /dev/null 2>&1 &' w$
fi
#END RASPIMJPEG SECTION

exit 0

So the code has been added correctly. 

My Raspbian version is Stretch, so could it be to do with the latest updates in stretch?

roberttidey, 2017-10-20:
I did an install with the September rasbian stretch and mine starts up Ok.

Your OS set up is either not executing the rc.local for some reason or some reports are that it is run earlier in the boot cycle. I am not quite sure how that could affect things.

One could put a printf or echo command in the rc.local to see if it was getting executed.
eg. echo 'ok' > /home/pi/rcok

rc.local is going out of fashion a bit now in favour of systemd so I may have to consider using that in the future.

hazardlabs, 2017-10-20:
I think I've found the fault

I had to add #!/bin/bash to the top of the file as it is blank on a fresh install of raspbian.

So now my script looks like:

#!/bin/bash
#START RASPIMJPEG SECTION
mkdir -p /dev/shm/mjpeg
chown www-data:www-data /dev/shm/mjpeg
chmod 777 /dev/shm/mjpeg
sleep 4;su -c 'raspimjpeg > /dev/null 2>&1 &' www-data
if [ -e /etc/debian_version ]; then
  sleep 4;su -c 'php /var/www/schedule.php > /dev/null 2>&1 &' www-data
else
  sleep 4;su -s '/bin/bash' -c 'php /var/www/schedule.php > /dev/null 2>&1 &' w$
fi
#END RASPIMJPEG SECTION

exit 0

And works on reboot. I hope this helps anyone else out

roberttidey, 2017-10-20:
OK. Thanks for that. 

I'll check further as the #!/bin/bash has been there for me

hazardlabs, 2017-10-20:
It may be because I'm using the raspberrypi-ua-netinst [https://github.com/FooDeas/raspberrypi-ua-netinst](url) for my host install as I've squeezed everything onto a 2GB SD card that I had flying around.

Whereas Raspbian lite still needed more than 2G.

Its still uses the Raspbian Repositories for the main installation so I would have assumed it would have been setup similarly.

Are you happy for me to close this issue, or wait until your findings? I'll be setting up another Rpi later today using the standard Raspbian Lite (Stretch) so can confirm the contents of the rc.local file then if you wish.

roberttidey, 2017-10-20:
Close if your check is OK.

This could be worth a note in wiki. Or I could maybe check in install if file is OK, but is probably better to move to systemd anyway. It is just thta is a bit more complicated to set up.

hazardlabs, 2017-10-20:
So in conclusion, a standard Raspbain(Stretch) install has a rc.local file with the following:

#!/bin/sh -e
#
# rc.local
#
# This script is executed at the end of each multiuser runlevel.
# Make sure that the script will "exit 0" on success or any other
# value on error.
#
# In order to enable or disable this script just change the execution
# bits.
#
# By default this script does nothing.

# Print the IP address
_IP=$(hostname -I) || true
if [ "$_IP" ]; then
  printf "My IP address is %s\n" "$_IP"
fi
exit 0

which includes the require #!/bin/sh -e line to mkae rc.local work correclty.

closed 2017-10-20 by hazardlabs