silvanmelchior/RPi_Cam_Web_Interface #377: Installing PHP fails on raspian 2017-09-07

nineclicks, 2017-10-23:
php5 is not found with raspian 2017-09-07, even after updating. I have to install php7 myself to get the interface working.

roberttidey, 2017-10-23:
The install.sh script offers a choice of whether to install php5 or php7.

I tried both on that raspbian version and they both seemed to work ok for me.

hazardlabs, 2017-10-23:
I can also confirm, that during my testing with the latest Raspbian Stretch image that php 5 doesn't work, having to re run the installer and set php7 in order for it to install and run successfully.

roberttidey, 2017-10-23:
OK. I'll try that again.

 Note that my test started with a perfectly clean raspbian (not NOOBS) and didn't have any php installed by default. If something installs php7 before the camera install.sh is run then that may confuse things when selecting php5 as the camera choice.

nineclicks, 2017-10-23:
I have been trying it on a fresh raspian, only first running an apt update and apt upgrade, installing vim, tmux and git, then enabling the camera in raspi config.