silvanmelchior/RPi_Cam_Web_Interface #378: Auto-Archive Feature

randuhmm, 2017-10-25:
I am interested in implementing a new feature to automatically "archive" files that are older than a specified number of days. I would like to have a scheduled task to move these files to an "archive" directory, which will really be a remote file system, to save space on my Pi file system. Thanks to user @roberttidey for pointing out that this functionality can be easily added by augmenting the existing "purge" functionality (#260).

I would also like to allow the browsing of these files, using the `preview.php` code. I have evaluated using the `preview.php` file to collect both files in the "media" directory as well as the "archive" directory, but the extensive use of the `MEDIA` define seems to create a problem there. It will likely require some overhaul on the `preview.php` page to get this working.

Here's an overview of what I am proposing for a feature-add:

1. Augment the "purge" feature from the `schedule.php` script to allow for "moving" of files rather than only "deleting".
2. Allow the `preview.php` script to browse both the "media" directory and the "archive" directory. This may require the `preview.php` script to check for and read the settings JSON file to locate the correct archive directory.
3. The destination "archive" directory shall be configurable in the `schedule.php` script.
4. The `preview.php` page should have functionality to manually "archive" a file. (stretch goal)
5. The `preview.php` page should have toggle button to show or hide files from the "archive" in addition to the "media" location.  (stretch goal)

So far, these are the primary features I have in mind, but I don't see this really being too difficult to implement in a week or two. PHP is no longer my primary language, but I am sure I can get familiar with it again to implement this feature.

Comments and/or questions are welcome and much appreciated!

roberttidey, 2017-10-25:
Sounds good.

I could probably help a bit on this one. E.g. changing the scheduler side

randuhmm, 2022-08-16:
@chadhutchins182 - Wow this is a feature I totally forgot about. I just checked my fork and it looks like I made some adjustments to the UI in the `www/schedule.php` file back in 2017. I only see one commit and it just seems to make some adjustments to the "Purge" functionality to enable the ability to archive rather than delete:

https://github.com/randuhmm/RPi_Cam_Web_Interface/commit/87c571de1a5e8fb6182d120306951b8fb745a495

I don't know how much more work is needed here to enable this feature, but I assume that the destination folder must be configurable and the logic for copy & delete (or just move) must be implemented.

chadhutchins182, 2022-08-16:
@randuhmm Is this the same/related as the #260 PR?

randuhmm, 2022-08-16:
> @randuhmm Is this the same/related as the #260 PR?

Yes, #260 doesn't actually implement the feature per se. It looks like it's adding a new isolated `preview.php` file for testing purposes. Ultimately, the application logic is unchanged from that PR. The work I was doing was to modify the `schedule.php` file which is responsible for the purge settings and functionality.

I may be able to pull the latest changes to my branch and drop it on a device for testing sometime this week. Or you are welcome to take ownership of the feature if you are interested in working on it.

chadhutchins182, 2022-08-16:
I was just trying to implement project itself but I'll take a look at your branch after I get it working first and see what I can do