silvanmelchior/RPi_Cam_Web_Interface #379: mmal_vc_component_create failed to create vc.ril.camera

Beschuitbus, 2017-11-07:
Hello,
installed the web interface on RPI-zero w.
When starting I get an error:
mmal: mmal_vc_component_create: failed to create component 'vc.ril.camera' (1:ENOMEM)
mmal: _component_create_core: could not create component 'vc.ril.camera' (1)
sh: 1: /var/www/html/macros/error_hard.sh: Permission denied

roberttidey, 2017-11-07:
Don't worry about the 3rd permission denied error message. That is just it trying to execute a macro script when it encounters an error.

The first two indicate that it is not managing to connect to the Raspberry camera.

Have you enabled camera support in the Raspbian OS? It is also worth checking using raspivid or raspistill to see if the basic camera is working OK.

Beschuitbus, 2017-11-08:
Hi Robert,
thanks for the quick responce, you're right, the camera cable is broken )-8

closed 2017-11-08 by Beschuitbus