silvanmelchior/RPi_Cam_Web_Interface #380: Wrong null path in the script

duchenpaul, 2017-11-18:
The NULL path is wrong in the script `RPi_Cam_Web_Interface_Installer.sh` line 113,114
Should be `/dev/null`, not `dev/null`

```bash
FN_STOP ()
{ # This is function stop
        sudo killall raspimjpeg
        sudo killall php
        sudo killall motion
        sudo service apache2 stop >dev/null 2>&1
        sudo service nginx stop >dev/null 2>&1
        dialog --title 'Stop message' --infobox 'Stopped.' 4 16 ; sleep 2
}
```

roberttidey, 2017-11-18:
Thanks. I have corrected that.

However, the recommended install method is now to use install.sh I don't maintain this old method which is just left in there.

duchenpaul, 2017-11-19:
Got it I think this ticket can be closed.

closed 2017-11-19 by roberttidey