silvanmelchior/RPi_Cam_Web_Interface #386: control from command line

Zacki84, 2018-01-07:
HI

I´m able to start/stop camera or videorecording or take a pic with the following command:
echo -n 'ca 0' >/var/www/html/FIFO
ru 0 or ru 1 => start/stop cam
ca 0 or ca 1 => start/stop rec
im => take pic

but when I try to change for example the Exposure Mode with 
echo -n 'em Night' >/var/www/html/FIFO
or
echo -n 'em Auto' >/var/www/html/FIFO

then I face an error message on the browser 

Why does this not work? I tried to change exposure while cam was streaming, and I tried to stop cam and change than, but its allways facing an error

Thanks

roberttidey, 2018-01-07:
Commands that use text values like em have parameters that are case sensitive.

The values that you see in the Web browser are just the labels not the values sent in the commands

So using 'night' and 'auto' will work

If you look in command reference (on scheduler page then it has the valid values for each command.

Zacki84, 2018-01-07:
thx, I got it

but I didn´t get Brightness running
echo -n 'br 60' >/var/www/html/FIFO
should be correct, but facing an error

roberttidey, 2018-01-07:
brightness (br) works on mine.

Note you do not need the -n anymore; the software was modified to handle the line feeds, but it will work even if you use the -n

First check the brightness works OK from web interface. Then do the command line version again and let me know what error you get.

Zacki84, 2018-01-09:
is working.

I insert a little break (time.sleep) in my python script between the EM line and the BR line, and its working now

Thanks

closed 2018-01-09 by Zacki84

roberttidey, 2018-01-09:
OK. If one does use LF between commands then the handler should accept those OK without worrying about pauses.