silvanmelchior/RPi_Cam_Web_Interface #390: RPi_Cam_Web_Interface

jwcd3m0cr3w, 2018-01-27:
Hi folks, is anyone still around,  to go back memory lane and help me out, real beginner here, but struggling and very keen to get this working.  
Chris

roberttidey, 2018-01-27:
What's the problem?

jwcd3m0cr3w, 2018-01-30:
Hi Guys,  apologies for this very amateur like question but being a beginner, and having spent a whole week and more, without success., I have had to come here for help, which will be much appreciated.

I have a RPi 3 model B and using Jessie stretch as the O/S, connected via LAN to my router, running apache 2, php7.0 and mysql.

I am trying to get RPi Cam Web Interface to work as described.

I have followed the installation instructions found here:
http://www.raspberry-pi-geek.com/Archive/2016/15/Using-the-RPi-Cam-Web-Interface

The problem I have is, the camera will not auto start on Boot. ( My biggest issue at the moment )

From the Web Browser ( Chrome ) when steered to the IP Address of my RPi, it shows the main page, with " LOADING " in bold letters and a few buttons.

**The camera has no power light**, not turned on, which is the reason why I guess.

It works fine if the camera is turned on manually from the config menu. 

Any assistance will be immensely appreciated.. I really need to get this working.

Many thanks

roberttidey, 2018-01-30:
Start up at boot is normally set up during the install when autostart is set to yes on the install options.

So you could try running the ./install.sh again and checking that option.

You can also check manually the following things.
1) autostart sets up the /etc/rc.local file to contain the code necessary to start the camera. Check this file (nano /etc/rc.local  It should contain a section starting 
#START RASPIMJPEG SECTION
and containing code that starts up the raspimjpeg process

2) The config file /etc/raspimjpeg contains all the run time options, The important one here is the autostart option near the end of the file. It should be set to standard for the system to work from boot.

roberttidey, 2018-01-30:
Couple of other items.

You say you have mysql. Have you loaded stuff after the RPi install that might have affected things?

rc.local should be enabled by default. You can check this by running the command `systemctl list-units | grep rc-local`
This should show it is active

jwcd3m0cr3w, 2018-01-31:
Hi roberttidey
Many thanks for your effort to help me out here..
I started from scratch this morning..
followed this installation from https://elinux.org/RPi-Cam-Web-Interface
-----------------------------------------------------------------------------------------------
cd RPi_Cam_Web_Interface

./install.sh
	Configuration Options:
	xCam subfolder 	html   ---  changed to /var/www/html -----
	xAutostart		yes
	xServer			apache
	xWebport		80
	Xuser			xx
	xPassword		xx
	xjpglink		no
	xphpversion	7
----------------------------------------------------------------------------
#This is edited config file for main installer. Put any extra options in here.
rpicamdir="/var/www/html/"
webserver="apache"
webport="80"
user=""
webpasswd=""
autostart="yes"
jpglink="no"
phpversion="7"
==========================================
pi@raspberrypi:~ $ sudo nano /etc/rc.local
======================================================
1.	autostart sets up the /etc/rc.local file to contain the code necessary to start the camera. Check this file (nano /etc/rc.local It should contain a section starting
2.	
#START RASPIMJPEG SECTION
and containing code that starts up the raspimjpeg process
3.	The config file /etc/raspimjpeg contains all the run time options, The important one here is the autostart option near the end of the file. It should be set to standard for the system to work from boot.
--------------------------------------------------------------------------------------------------------------
**# pi@raspberrypi:~ $ sudo nano /etc/rc.local**

#!/bin/sh -e
# rc.local
# This script is executed at the end of each multiuser runlevel. ~
# Value on error.
## In order to enable or disable this script just change the execution
# bits.
#
# By default this script does nothing.
# Print the IP address
_IP=$(hostname -I) || true
if [ "$_IP" ]; then
  printf "My IP address is %s\n" "$_IP"
fi
#
#Auto start Tightvncserver:
su - pi -c '/usr/bin/tightvncserver :1'

**#START RASPIMJPEG SECTION**

mkdir -p /dev/shm/mjpeg
chown www-data:www-data /dev/shm/mjpeg
chmod 777 /dev/shm/mjpeg
sleep 4;su -c 'raspimjpeg > /dev/null 2>&1 &' www-data

if [ -e /etc/debian_version ]; then
  sleep 4;su -c 'php /var/www//var/www/html//schedule.php > /dev/null 2>&1 &' w$
else
  sleep 4;su -s '/bin/bash' -c 'php /var/www//var/www/html//schedule.php > /dev$
fi

#END RASPIMJPEG SECTION

################################
#  Config File for raspimjpeg  #
################################

# Autostart
#
# autostart: standard/idle
# motion detection can only be true if autostart is standard
#
autostart standard
motion_detection false
---------------------------------------------------------------------
systemctl list-units | grep rc-local

This should show it is active

What I get is below:

pi@raspberrypi:~/RPi_Cam_Web_Interface $ systemctl list-units | grep rc-local
● rc-local.service      loaded failed failed    /etc/rc.local Compatibility     

This is the best info that I can provide at this time....      

----------------------------------------------------------------------------------

roberttidey, 2018-01-31:
OK. The basic problem is that for some reason the rc.local service has a problem with your overall set up.

This is not directly related to the camera software but this service allows boot time scripts to run which is what starts up the camera system.

When you say you started from scratch was that from a clean OS image before loading anything else. Reason for asking is that something else might be causing rc.local a problem. If you could try just loading the camera software onto raspbian before doing anything else that might help trace the problem.

I will also check out what could cause rc.local to fail.

jwcd3m0cr3w, 2018-01-31:
I configured a new .img file with pre-configured - things like SSH - Camera Enabled - Local config etc , and installed apache - php and mysql - 
This now becomes my .img file with everything upto apache, php and mysql ( Mariadb ) installed and working good.
Nothing else has been changed..  
When I boot, I can test index.html and index.php in my web browser fine -- ok.

**My rc.local file looks lie this:**

#!/bin/sh -e
# rc.local
# This script is executed at the end of each multiuser runlevel.
# value on error.
# In order to enable or disable this script just change the execution
# bits.
# By default this script does nothing.
# Print the IP address
_IP=$(hostname -I) || true
if [ "$_IP" ]; then
  printf "My IP address is %s\n" "$_IP"
fi
#
#Auto start Tightvncserver:
su - pi -c '/usr/bin/tightvncserver :1'
#
**#START RASPIMJPEG SECTION**
mkdir -p /dev/shm/mjpeg
chown www-data:www-data /dev/shm/mjpeg
chmod 777 /dev/shm/mjpeg
sleep 4;su -c 'raspimjpeg > /dev/null 2>&1 &' www-data
if [ -e /etc/debian_version ]; then
  sleep 4;su -c 'php /var/www//var/www/html//schedule.php > /dev/null 2>&1 &' w$
else
  sleep 4;su -s '/bin/bash' -c 'php /var/www//var/www/html//schedule.php > /dev$
fi
**#END RASPIMJPEG SECTION**

jwcd3m0cr3w, 2018-01-31:
I have prepared several pre-configured .img files. 
The one I am using here,  is Jessie-stretch with apache2, php7.0 and mysql pre-installed and tested.

My rc.local file looks like this:

#!/bin/sh -e
#
# rc.local
#
# This script is executed at the end of each multiuser runlevel.
# value on error.
#
# In order to enable or disable this script just change the execution
# bits.
#
# By default this script does nothing.

# Print the IP address
_IP=$(hostname -I) || true
if [ "$_IP" ]; then
  printf "My IP address is %s\n" "$_IP"
fi
#
#Auto start Tightvncserver: - This starts my VNC desktop server on boot - works fine.
su - pi -c '/usr/bin/tightvncserver :1'
#
#START RASPIMJPEG SECTION  -- No Idea what is going on in here "

mkdir -p /dev/shm/mjpeg
chown www-data:www-data /dev/shm/mjpeg
chmod 777 /dev/shm/mjpeg
sleep 4;su -c 'raspimjpeg > /dev/null 2>&1 &' www-data
if [ -e /etc/debian_version ]; then
  sleep 4;su -c 'php /var/www//var/www/html//schedule.php > /dev/null 2>&1 &' w$
else
  sleep 4;su -s '/bin/bash' -c 'php /var/www//var/www/html//schedule.php > /dev$
fi
#END RASPIMJPEG SECTION

exit 0
=========================================

I will do as you suggested - try out the RPI_Cam_Web_Interface before installing Apache etc,
tomorrow and will keep you posted -  Bur it does ask for an installed server on setup- ? what should I do at that step ?

roberttidey, 2018-01-31:
It installs everything it needs including apache and php. Pre-installing stuff can sometimes confuse matters. If you are using Apache for other stuff then any conflicts need to be addressed.

It is difficult to work out your rc.local file with the formatting going on but I am guessing a lot of those lines in bold at the beginning actually have a # in front to comment them out.

I am also suspicious about the tightvncserver line. It could be that it never returns from that and therefore never executes the RASPIMJPEG part. This would explain why it doesn't start up on boot.

Try commenting out the tightserver line with a # at the front.

If that fixes it then you need a way to start the tightvnc server in the background or you could try moving it after the RASPIMJPEG section.

jwcd3m0cr3w, 2018-02-01:
Hi roberttidey
I have very good news.. As you suggested, I started with a new .img file, without apache, php,mysql pre-installed. .
Installed RPi_Cam_Web_Interface --- Bob's your Uncle   --  It bleeding well worked...
Rebooted several times and every time it worked fine.
I then used Jessie_stretch_lite on my pi0 W -- which is the machine I want to use for this project, followed the installation instructions, and WOW -                                it worked.
whilst installing the software, i saw this,   Configuration Options:
	xCam subfolder 	html
At this moment, I am trying to locate the above... Will want to modify the appearance of the home page

Many thanks for responding to my request for assistance..  My initial goal has been reached.

I will now fiddle with Camera settings etc and will keep you posted
On the pi 0 - with Jessie_stretch_lite - there is no desktop...  Putty works fine.. No need for tightvncserver or xrdp.. 
On the pi 3 - took the tightvncserver commented out start up on boot -     it still works...

Many thanks.. Sincerely appreciate it... I have put together two text files, something like an IDIOTS GUIDE to get this working first time..  Let me know if these would be useful to you and your team..

Chris

jwcd3m0cr3w, 2018-02-01:
Hi roberttidey

Your help was absolutely fantastic..   I now have a Pi 3 LAN connected, and another pi 0 W - wireless connection. Both have camera's installed and work like a dream..  Gotta go now..
Will keep you posted.
Chris

roberttidey, 2018-02-02:
Good to hear its working now.

The web software is installed in /var/www/installsubfolder so if you leave it at default this is at /var/www/html

You can also leave the subfolder blank during install in which case it is just /var/www This can be convenient if you have no other web stuff running as the camera access then becomes just http://cameraip

There are other ways of starting tightvncserver up at boot if you need that. These would probably avoid any conflict.

You can add / edit the wiki at eLinux. For example by including any extra notes under tutorials section. You just need to create a elinux login if you haven't got one. (The team is just me!)

jwcd3m0cr3w, 2018-02-04:
Hi roberttidey

Many thanks for your continued support..  Need some more advice., please.
Everything works fine without setting security on initial install... I No user id and no password.

I then decided to introduce this security by setting a user id and password.
pi@raspberrypi:~/RPi_Cam_Web_Interface $  ./install.sh

Now, I do NOT see the default camera page :: Just a page as shown below;;
  No Login pop up window appears, just,,,

**Index of /**
  | Name | Last modified | Size | Description
-- | -- | -- | -- | --
Apache/2.4.25 (Raspbian) Server at 192.168.1.130 Port 80

If the security is removed, everything works normally again..

Do not have a clue as to what is happening...  

As always,, guidance and assistance sincerely appreciated.

roberttidey, 2018-02-04:
Normally it just works.

Assuming you have the subfolder at html then http://192.168.1.30/html should bring up a user / login pop up window. If in any other subfolder then url changes to match.

First, I assume you do not have a pop-up blocker or setting on the browser. If possible try another browser.

Second when install.sh sees a user / password then it does the following (assuming apache)

1) runs the htpasswd command to create a security file /usr/local/.htpasswd

2) creates a .htaccess file in the www folder (e.g. /var/www/html. This contains the authentication parameters and points to the .htpasswd file in /usr/local for password checking.

3) Sets the apache2 conf file (/etc/apache2/sites-enabled/raspicam.conf) to have a setting 'AllowOverride All' This is what triggers the password dialog.

Check those all exist OK. It is possible if you are using apache for other things that they have changed how this works.

jwcd3m0cr3w, 2018-02-04:
Hi roberttidey,
Many thanks, did not have time to go through your advice as yet, but rest assured I will work on this and will keep you informed... 
I am very interested in the raspberry pi and would like to complete a project, using a pi 0 / pi 3 - wireless or LAN, connected, being able to stream video on local network and be accessible  via the internet.
You have helped me get pretty far into this project, I want to learn as I move forward and I really appreciate your help.  Maybe one day I will be able to help newbies..  Will get back soon.
Cheers

jwcd3m0cr3w, 2018-02-04:
Before I retire for the night, I thought I should share some good news with you. 
I have apache installed.  On my Local LAN, I have a wired pi 3 - a Win 7 PC - Wireless - an apple Ipad - a mobile phone & a pi 0  W ( currently switched off )  == Here is what I have got working  without any security  settings - 
**The pi 3 turns the camera ON**   this camera can now be accessed from all the  other devices  - that;s fantastic, I have discovered a few pit falls to this system, which I will test and keep you informed.
I want to actually understand what is going on rather than just getting something working without at least some understanding of what is going on.
Thanks for all your assistance..  Will keep you posted.

jwcd3m0cr3w, 2018-02-06:
Hi, hope you are well ?  Hope I am not offending you in any way.  I just want to keep you informed..
Progressing from my earlier update,
I set up apache security   "  User / Password --- pi / ABCD  "  - re-booted -- tested access to the camera from all the other 3 devices, local LAN, and Yes the login screen appeared.
It worked with every one logging in as "  pi / ABCD  "  ....  Each device had visibility of the camera output, and also had access, to the main control buttons, such as the, Start and Stop,  camera button.
Each device has something like " admin " access.  I was thinking of having a multiple user / pwd  register, with users such as ( guest, tim, john, mike, etc ) have access to the camera live stream only. no access to the control panel'
Others , like  ( admin ), will have full access to the camera and system settings .
Hoping you can guide me in getting my little project moving further...
Your assistance and guidance, as has been said before,  is immensely appreciated
Many thanks.

roberttidey, 2018-02-07:
BY default there is one login with full access as you describe.

However, there is built in the option to have multiple users with 3 levels of access (full, basic controls, and preview only). This is described in the wiki under additions and tricks 'User access levels'

You have to add user/logins to apache using standard methods and then create a text file on the server allocating user names to access levels.