silvanmelchior/RPi_Cam_Web_Interface #391: connection to camera

AhmadWahid, 2018-02-05:
hello i'm new in raspberry , and i have install Rpi Web Cam interface , I want to know , how this project run , how html can show streaming from camera , and how to make record button and take picture button , anyone can help me to explain this project, and how to built it

roberttidey, 2018-02-06:
First read through the wiki to understand the functionality of the RPi_Cam system. As installed it provides a web based access to the camera system giving controls over recording and capturing, motion detection and a scheduling system.

It has two basic parts.
The first is a process called raspimjpeg which is running in the background. The source for this is linked in the wiki. It consists of files called RaspiM* in the userland\host_applications\linux\apps\raspicam folder. Instructions for building a new version are in the readme, but the most recent compiled version is included in the overall install. raspimjpeg opens connections to the camera via its MMAL interface and this lets it get detailed control over all aspects of the camera. It extracts preview data from the camera and composes this into a sequence for preview presentaion as a mjpeg ream.. I also accepts commands via a PIPE interface and this allows it to adjust settings, take images, record videos, motion detect etc. Details of the commands are in the Wiki. The PIPE mechanism also allows any other program (scripts, python, other web view) to get simple access to the raspimjpeg functionality.

The second part of the software is the web server and client files. This is based around a normal web server (usually apache). It interacts with raspimjpeg to provide a streaming preview and issues commands via the PIPE interface. The presentation of the captures taken and the whole scheduling system is run by php files under the web servers control.

AhmadWahid, 2018-02-06:
Thankyou sir , can you give me link for tutorial , so i can directly go to
the page

roberttidey, 2018-02-06:
The information available is the
wiki at  https://elinux.org/RPi-Cam-Web-Interface
RPI cam web software and installer in this github
raspimjpeg github at https://github.com/roberttidey/userland

If you have specific questions ask them here.

fantyd, 2018-02-06:
I have problem installing RPi Cam Web Interface, it will show the below: anyone can help?

pi@raspberrypi:~/cam $ ./install.sh
Reading package lists... Done
Building dependency tree
Reading state information... Done
Package dialog is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source
E: Package 'dialog' has no installation candidate

AhmadWahid, 2018-02-06:
I'll call you soon , thankyou so much sir

Pada tanggal 6 Feb 2018 6.28 PM, "roberttidey" <notifications@github.com>
menulis:

The information available is the
wiki at https://elinux.org/RPi-Cam-Web-Interface
RPI cam web software and installer in this github
raspimjpeg github at https://github.com/roberttidey/userland

If you have specific questions ask them here.

fantyd, 2018-02-06:
Hi are you located in Singapore?

fantyd, 2018-02-06:
Hi, i can't install RPi cam web interface, can i check with you ?

AhmadWahid, 2018-02-07:
No , i'm in Indonesia

AhmadWahid, 2018-02-07:
Have you see video ini YouTube

fantyd, 2018-02-07:
Hi, I managed to solve my aptget upgrade issue by using other network, so I believe it could be cause by my home network and I would try install the Rpi cam again and see how. Thanks!!

AhmadWahid, 2018-02-08:
Good luck with it , ini my case , first time I install this software i
failed too, then i copy this software to my repository , and install it
again use my repository link

Pada tanggal 7 Feb 2018 4.56 PM, "fantyd" <notifications@github.com>
menulis:

Hi, I managed to solve my aptget upgrade issue by using other network, so I
believe it could be cause by my home network and I would try install the
Rpi cam again and see how. Thanks!!

AhmadWahid, 2018-02-10:
Hello sir , i almost read all content on Wiki , but i haven't see how to
make video recording button , could you tell me how to make it sir?

roberttidey, 2018-02-10:
I assume you have the normal client running OK with all the different capture buttons.

This works by having functions associated with the buttons that then send commands to the server. These commands are sent using a URL which puts the cmd into the raspimjpeg process. So for example (assuming software installed into html subfolder) http://host/html/cmd_pipe.php?cmd=im puts the command im in which then takes an image. For video recording one uses the ca command. ca 1 starts a video recording and ca 0 stops it. So http://host/html/cmd_pipe.php?cmd=ca%201 starts a recording and  http://host/html/cmd_pipe.php?cmd=ca%200 stops it. The %20 in these is a space character.

You can test these URLS by having the normal client running and then browsing to the URLs in another tab. You should see the recording start and stop.

AhmadWahid, 2018-02-12:
hai sir , can not understand it , can you tell me how to build this pcoject from start, what should i do , i try to read your HTML code , but i can't understand ,

roberttidey, 2018-02-12:
I am not sure what you are trying to do.

To use the camera system you do not have to build it.

If you are trying to modify the layout and functionality of the web based part then you can either modify the code that exists (e.g. index.php) or you can create a new html page that uses the same command mechanism to communicate with the server.

With a php web server system, any php that is requested from the server like index.php is first executed by the php interpreter and then the resulting html is delivered back to the browser.

One way you can see this is to do a 'view page source' of the main page (right click on the page). This is the html that is delivered back to the browser when the index.php page is requested. You can save this and compare it with index.php to see how it was generated.

AhmadWahid, 2018-02-13:
I have an Exam sir , and i want to make same projects with this Cam
interface , I want to learn how to make this project, can you help me sir ,

AhmadWahid, 2018-02-13:
I'm sorry if i ask to much sir, i just want to know and learn more about
your project,

roberttidey, 2018-02-13:
I think maybe what you need to do is get familiar with web/php programming then look again at how the index.php delivers the html to the browser and the functions that it then provides.

There are lots of tutorials on these subjects. Try https://www.w3schools.com/php/php_intro.asp

If you have some specific question on how part of this camera software works then I will try to answer it

AhmadWahid, 2018-02-13:
Thanks sir , i'll do it , but can you explain to me how you connect camera
to HTML page ,

roberttidey, 2018-02-13:
There is a c program called raspimjpeg. THis is running all the time and connects to the camera via an interface called MMAL.

This interface allows program control of the camera and collecting data from it in real time.

One of the things raspimjpeg is doing is requesting the camera to give it a feed of snapshot images at a reduced resolution. These are placed in a temporary folder at /dev/shm/mjpeg and if you look there you will find a cam.jpg which is a continuously updated image from the camera. It is typically updated 25 times per second.

The web interface can now produce a view on the screen by continuously requesting an update of this image. Ajax is used to perform this request. A better method is also supported called MJPEG streaming. In this case the web browser makes one request and then gets back the sequence of images as a single stream which updates the screen.

AhmadWahid, 2018-02-13:
MMAL what is it?is that some kind of program language?

wingers999, 2018-02-13:
What is MMAL

[http://lmgtfy.com/?q=mmal](url)

wingers999, 2018-02-13:
What is MMAL

[](http://lmgtfy.com/?q=mmal)

wingers999, 2018-02-13:
What is MMAL?

[lmgtfy.com/?q=mmal](url)

wingers999, 2018-02-13:
http://lmgtfy.com/?q=mmal

AhmadWahid, 2018-02-13:
How rasppimjpeg can run in background, and open every raspberry turn ON,
and what is function from command inside install.sh

roberttidey, 2018-03-08:
For multiple cameras there is a page that can show multiple previews

Look up multiview in wiki

AhmadWahid, 2018-05-23:
Hello i have installed Raspimjpeg on my RASPBERRY , and i have question
"how i Van change path for image and video, i have read on website but , i
can't do it, cantik you give me example for it , i want to change the
storage into my computers using lan cabel

AhmadWahid, 2018-06-03:
> First the new location needs to be accessible to the Raspberry Pi. If this
> is an external disk drive or network connection then you need to mount it.
> This is achieved by adding an entry into fstab so that it is automatically
> mounted on boot.
>
> There are many guides on how to do this on the web. One example is
> http://www.noveldevices.co.uk/rp-cifsinfstab
>
> Once you have done this then you need to set the permissions on the
> mounted drive. You can either change the ownership to www-data using chown
> or just set the overall permissions to 777 using chmod
>
> You can now link the mounted drive to the media folder using a command like
>
> sudo ln -sf /mnt/mydrive /var/www/html/media
>
> where /mnt/mydrive is where your network share is mounted and this then
> appears at the normal media localtion assuming you used html as the default
> folder to install into.

i have done with 1st step , but i can't chown /mnt/folder that i have
linked, i have use , chmod -R 777, and chown , but acces denied , do you
have any suggestion?

AhmadWahid, 2018-06-03:
i have link /var/www/camera(my raspimjpeg folder)/media to my PC shared
folder it's name /var but i can't take any pictures or videos , am i miss a
step or something?

2018-06-03 12:30 GMT+07:00 AhmadWahid XIIRPL_1 <ahmadwahidxiirpl1@gmail.com>
:

roberttidey, 2018-06-03:
You need to use sudo in front of chown and chmod commands.

After the link you should then be able to ls -l /var/www/camera/media and see files in the shared folder.

Providing the permissions are ok then the camera software should store its media data in the shared folder. If not then check scheduler log for any errors.

AhmadWahid, 2018-06-04:
i have use sudo in front chown and chmod , but it's denied ,
i use "Sudo Chown pi /mnt/save" and i try use "Sudo chmod -r 777 /mnt/save"

AhmadWahid, 2018-06-04:
i can do lb -l /var/www/camera/media , but i can't take any videos or
pictures from web interface , when i click on the video button nothing
happend

2018-06-04 9:52 GMT+07.00, AhmadWahid XIIRPL_1 <ahmadwahidxiirpl1@gmail.com>:

AhmadWahid, 2018-06-04:
okay sir , i have solved it , I link my windows folder directly to
/var/www/camera/media , it's work well, Thankyou very much

2018-06-04 10:34 GMT+07.00, AhmadWahid XIIRPL_1 <ahmadwahidxiirpl1@gmail.com>:

AhmadWahid, 2018-06-04:
i have other question sir , i set my ip to use my raspberry directly
using LAN cabel , i use it to connect shared folder windows to
raspberry , and i want to remote it from internet use wifi ,is that
possible to do ?

2018-06-04 10:37 GMT+07.00, AhmadWahid XIIRPL_1 <ahmadwahidxiirpl1@gmail.com>:

roberttidey, 2018-06-04:
Yes. It is possible to access the web interface from the internet. This is covered in a section on the wiki under additions and tricks.

Basically you need to set up port forwarding on your router so that a a request to a particular port from the internet is forwarded onto the raspberry pi. There is a link in that section on the wiki that helps with that.

As access from the internet potentially allows anyone to try to access the camera then it is normally a good idea to set up user/password access as well to provide security.

AhmadWahid, 2018-06-04:
Okay Mr.Robert , I'm so glad to do it soon , thank you very much 👍