silvanmelchior/RPi_Cam_Web_Interface #392: RPi Cam Web Interface installing

fantyd, 2018-02-06:
I have problem installing RPi Cam Web Interface, it will show the below, anyone can help?
pi@raspberrypi:~/cam $ ./install.sh
Reading package lists... Done
Building dependency tree
Reading state information... Done
Package dialog is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

E: Package 'dialog' has no installation candidate

roberttidey, 2018-02-06:
dialog is used to allow the install script to present little menus. So it is installed first.

Sometimes package install from the repositories can fail due to temporary connection issues at the raspberry pi end.

It seemed to be available when I gave it a try just now

Give it another try. You can also try running sudo apt-get install dialog from the command line.

fantyd, 2018-02-06:
do i need to retype the step below again?
git clone https://github.com/silvanmelchior/RPi_Cam_Web_Interface.git
cd RPi_Cam_Web_Interface
./install.sh

fantyd, 2018-02-06:
i keep getting the error below (Hash Sum mismatch)

Fetched 6,327 B in 3s (1,698 B/s)
E: Failed to fetch http://mirror.nus.edu.sg/raspbian/raspbian/pool/main/d/dialog/dialog_1.3-20160828-2_armhf.deb  Hash Sum mismatch
   Hashes of expected file:
    - SHA256:63d314f73c2424c0dfdc297351d67537b452836920744dd257aba99eca7d5143
    - SHA1:897e00d45fa939b8c8d5da2e48695ea39b8ce13d [weak]
    - MD5Sum:8d34d6745ef18594f1a190f1ff22d5ba [weak]
    - Filesize:239190 [weak]
   Hashes of received file:
    - SHA256:c35668cb0b91b3117291a34ed7e97b4242ee44340e95101f29c0cbd996227621
    - SHA1:2cee863997a9f3f654e70dd321870694882fab45 [weak]
    - MD5Sum:b41606b7e78eea67e2a108c5f734a17b [weak]
    - Filesize:239190 [weak]
   Last modification reported: Sat, 03 Dec 2016 10:08:07 +0000
E: Unable to fetch some archives, maybe run apt-get update or try with --fix-missing?

roberttidey, 2018-02-06:
Have you run sudo apt-get update and sudo apt-get dist-upgrade first?

Those errors normally indicate that your OS is not up to date with the repositories. It could be the package download is failing and so giving a bad checksum, but that will normally go away if you retry

fantyd, 2018-02-06:
My upgrade also have this error, I tried many method and none of them work.

fantyd, 2018-02-06:
I tried reinstall many time of Raspbian os and update then upgrade, error keep occur.

pishotta, 2018-02-06:
Hi Fantyd,

 

Like Robert says, this looks like a problem with the Raspian update procedure.  Sometimes there is a problem with the repositories that goes away after a day or so.  In any case, it doesn’t appear to have anything to do with the RPI Webcam software package.

 

You might try searching/posting your problem on the forum here:  https://www.raspberrypi.org/forums/ , where they discuss such problems.

 

Good luck,

 

-- Fred P.

fantyd, 2018-02-07:
Hi pishotta,

I solved the error by using other network, the problem might be cause by ISP, modem or router, I will try to install the software and see how. Thanks for the help!

fantyd, 2018-02-07:
Hi pishotta,

Can I ask you do RPi cam app have the feature of email alert when motion detected?

roberttidey, 2018-02-07:
It doesn't have email alert directly built in but if you check the wiki out you will see that this can be done using the job macros facility.

I think there were posts in the forum describing this in more detail.

fantyd, 2018-02-07:
Hi roberttidey,

can i ask more in detail, the wiki indicate the script need to change as below, so there is a feature in the cam interface of editing the script?

#!/bin/bash
list=( $1*.th.jpg )
thumb="${list[-1]}"

roberttidey, 2018-02-07:
The web interface allows you to enable/disable macros and to choose which macro script to associate with each action but not to edit the scripts themselves.

You create new macro scripts with any text editor and then put them in the macros subfolder. They must be given ownership by www-data and given execution permissions as described in the wiki.

For example to create a motion_event.sh macro I would copy the existing test.sh and change ownership and permissions.

cd /var/www/html/macros
sudo cp test.sh motion_event.sh
sudo chmod +x motion_event.sh
sudo chown www-data:www-data motion_event.sh

Now you can edit motion_event.sh and replace the actions by whatever you want (e.g. email).

Also read the tutorial referred to in the wiki under 'Telegram Alerts'
https://quavoce.wordpress.com/2017/09/29/telegram-send-messages-photos-animated-gifs-from-your-raspberry-pi-and-rpi-web-cam/

This gives a description of creating notifications (not email) but describes the macro creation process.

fantyd, 2018-02-07:
The alert is not email alert but a message alert via a app, is there a step by step guide to create email alert?

roberttidey, 2018-02-07:
You just need to use ssmtp to send the mail. Search for raspberry ssmtp send email. Get it working from the command line first then put into a script.

fantyd, 2018-02-08:
Ok, thanks!

closed 2018-02-09 by fantyd