silvanmelchior/RPi_Cam_Web_Interface #395: button function in RPi interface not working

fantyd, 2018-02-09:
The RPi interface is up, but record video, record image, motion detection start and selected motion detect mode to internal, all seem not functioning, do i miss any step?

roberttidey, 2018-02-09:
If the interface is working, are you getting a live preview of the camera output?

If not then the raspimjpeg process is likely to be not activated. Did you tell it to autostart on boot when you installed. If so then it should start up when you boot up.

You can also manually start it by running the ./start.sh script in the RPi install folder.

fantyd, 2018-02-09:
Yes, the live preview is up too.

fantyd, 2018-02-09:
where is the RPi install folder?

roberttidey, 2018-02-09:
That folder is the one you ran ./install from.

If live preview is working then raspimjpeg process is running OK.

Are the buttons disabled or just not working?

Check the status file by 

cat /var/www/html/status_mjpeg.txt

It should show ready

Note you may need to change the html in that command if you have installed in a different folder.

If it shows ready then there is something going wrong at your browser. Try turning on developer tools in the browser (F12) as that will show errors occuring there.

fantyd, 2018-02-09:
I run ./start.sh in the command line, afterthat, still the same, nothing happen on clicking the button

fantyd, 2018-02-09:
it is showing ready but what you mean "it should show ready" and if it show ready then something going wrong?

fantyd, 2018-02-09:
i using google chrome, would it work on RPi interface?

roberttidey, 2018-02-09:
When you run that command then you should get back the word 'ready' as a response

I use Chrome all the time. The developers tools allow you to see what is going on.

fantyd, 2018-02-09:
ok, i in the developers tools, pls let me know where to see any issue?

roberttidey, 2018-02-09:
When the buttons are active then they should be bright colours, when they are inactive then the colour is dimmed.

Are you getting the word 'ready back on the command line when you run that command?

In developers tools on the network tab you should see a stream of files being fetched. Every few seconds there should be a status_mjpeg.php?last=ready That is the server telling the browser that the system is ready. If so then the buttons should be activated.

If the button is active then when you press the record image button then you should see cmd_pipe.php?cmd=im  being sent which is the browser telling the server to take an image

fantyd, 2018-02-09:
it is show the next line "readypi@raspberrypi:~/RPi_Cam_Web_Interface $" using the command below. and t
cat /var/www/html/status_mjpeg.txt
and the button is review as bright colours.

fantyd, 2018-02-09:
i clicked many times, it just keep showing "cam_pic.php" all the way

fantyd, 2018-02-09:
Thus, what could I do next?

roberttidey, 2018-02-09:
The cam_pic.php repeating is normal. That is it fetching the images to build the preview screen but you should see the other commands in there.

You can get rid of the stream of cam_pic.php by clicking on the System link on the web interface and then clicking on MJPEG-Stream button. That allows the preview to be sent as one continuous stream. That makes it easier to see the other commands.

When you press on the Record Image button does it go dim for a second or so and then return bright. That is it taking an image. Have you clicked on the Download Videos and Images button? That is where captures are shown.

fantyd, 2018-02-09:
I just tested one more time, it is show md_pipe.php?cmd=im, but the status is showing (canceled) and when i clicked on the download image and video, nothing is record.

fantyd, 2018-02-09:
it is the same happen in using internet explorer and mozilla firefox

fantyd, 2018-02-09:
When i clicked the Record button, it will remain as dim all the way

roberttidey, 2018-02-09:
There is no status called cancelled so I am puzzled by that.

From a command line do the following cmd
echo 'im' > /var/www/html/FIFO

That will cause the camera system to take a picture. You should see the jpg of the picture and a thumbnail in the /var/www/html/media folder. It should also show up in the web browser under Download Videos.

Can I check, which folder did you install into; html ? Are you browsing to http://cameraip/html where cameraip is the ip address of the raspberry?

roberttidey, 2018-02-09:
The other strategy here is to start again.
Clean Raspbian OS
Enable camera
Check that raspistill takes a picture to make sure camera is working.
Do NOT install anything else
Follow the eLinux wiki install instructions.
It will just work

fantyd, 2018-02-09:
I just type http://ip of the raspberry, then it show the interface

fantyd, 2018-02-09:
I will try and let you know again

fantyd, 2018-02-09:
But the raspbian os image, does it need to be full desktop version or the minimum based version? Or both can work

roberttidey, 2018-02-09:
So when you installed did you leave the subfolder as html or did you blank it out?

By default it is html, the software gets put into /var/www/html, and you should browse to ip/html

If you blank it out then it gets put in /var/www and you should browse to ip/

It sounds like you might have done installed into both places so the raspimjpeg process is running thinking it is under html but you are accessing it from /var/www. This would mean the status and the commands would be going to the wrong place and not working.

Try browsing to ip/html if you installed there.

roberttidey, 2018-02-09:
Both full and Lite will work, but I prefer to install Full and then remove unnecessary stuff as described in the wiki.

fantyd, 2018-02-09:
Which part of it show unnecessary stuff?

roberttidey, 2018-02-09:
Installation tips

fantyd, 2018-02-10:
Everything work well after os reinstall, remove unnecessary stuff and RPi installed. Thanks! By the way, can I go to the folder that save the record image to see directly rather than download in the interface?

roberttidey, 2018-02-10:
Good to hear that.

Yes. The files in the media folder can be accessed by other programs.

fantyd, 2018-02-10:
U mean those record image can found in the path “ /var/www folder“? Is the path right?

roberttidey, 2018-02-10:
There is a media folder in the location where the web server is installed. So if you installed with a subfolder of html (default) then the path would be /var/www/html/media. If you had blanked out the subfolder during the install then the path would be /var/www/media

The media folder is where all recordings and thumbnails are stored,

fantyd, 2018-02-10:
Ok thanks!

fantyd, 2018-02-11:
Can I ask it seem that if device using the same network by login RPi interface using raspberry pi IP address, in this case, anyone can open the interface, is there an authentication or password to login?

roberttidey, 2018-02-11:
Look in wiki under Additions and tricks 4.3 and 4.15

4.3 describes the install time set up of username / password which will apply to all users accessing the camera. You can just run the ./install.sh script again if it has already been installed.

4.15 describes a multi-user method where different users are granted different access levels when they log in.

fantyd, 2018-02-11:
Ok thanks !

fantyd, 2018-02-12:
Is the steps involve router, can I don’t involve router and set user login, if there is, is there a step by step guide?

roberttidey, 2018-02-12:
Setting a username / password for the interface does not need a router change when accessing the camera from the local network.

If you want to access the camera from outside your local network (e.g. from internet) then you do need to change the router settings to port forward to the camera when accessed from outside. The user / password setup remains unchanged and will apply for the external access. There are lots of on-line guides on how to set up port forwarding.

fantyd, 2018-02-12:
I have already setup the username and password and enable it using the command ./install.sh, after system reboot up, it does not prompt me for the username and password to login to RPi interface?

roberttidey, 2018-02-13:
The username password normally just works. You must have something strange in what you have done like installing into 2 places to stop this.

I suggest you remove all the files under /var/www and then run the ./install.sh again with html subfolder username and password set up.

You can remove all the files by
cd /var/www
sudo rm -rf *

fantyd, 2018-02-13:
if remove, would it remove other folder "/html/macros/"?

fantyd, 2018-02-13:
before i remove, i realize i can't see /var/ folder in file manager?

fantyd, 2018-02-13:
the password is working now, but why I can't see /var/ folder in the file manager?

roberttidey, 2018-02-13:
Good to hear password is working.

File manager may be hiding this from a normal user. If you use a terminal and type ls /var then you will see it.

fantyd, 2018-02-13:
How can I do to see var folder in the file manager?

roberttidey, 2018-02-13:
I can see the var folder in file manager without problems. If you start with root you should see the var folder.

If not then I suggest you ask a question in the standard raspberry pi forums as this is not really a RPi Cam question.

fantyd, 2018-02-14:
Ok sure thanks!