silvanmelchior/RPi_Cam_Web_Interface #396: Setting the time

modsbyus, 2018-02-12:
Is there anyway to add the feature of updating the time on the Pi through the web interface? I am using this as a wifi ad-hoc camera and would like to be able to update the system time if necessary from the web interface.

roberttidey, 2018-02-12:
Yes. That should be possible.

I assume this is for a case where Pi (ad Hoc) can't sync its time via the internet.

The main work would be on the user interface side. If it was kept simple by entering a compatible date string (e.g. "12 FEB 2018 23:00:00") then that would make it a lot easier.

I'll check that out.

modsbyus, 2018-02-12:
Thanks!! I look forward to hearing back from you.

roberttidey, 2018-02-13:
Latest version should now support this. Under system settings you can edit the date / time string and press OK.

It will have no effect if normal internet NTP is being used. I had to turn my NTP off to test 
sudo timedatectl set-ntp 0

modsbyus, 2018-02-13:
Wow!! That was fast! I will check it out this evening.

modsbyus, 2018-02-13:
I had an oportunity to look at this earlier than I expected. I see the change but I am seeing this error in the CLI...
```
>>> /etc/sudoers.d/RPI_Cam_Web_Interface: syntax error near line 5 <<<
sudo: parse error in /etc/sudoers.d/RPI_Cam_Web_Interface near line 5
sudo: no valid sudoers sources found, quitting
sudo: unable to initialize policy plugin
```

roberttidey, 2018-02-13:
Strange. Looks like an extra line feed may have crept in somewhere.

I've updated it. Try again.

modsbyus, 2018-02-13:
I ran update.sh from the Rpi cam directory and it updated but I am still seeing that error. It printed that error 5 times immediately following the update.

roberttidey, 2018-02-13:
OK I am checking it out further. It is the file I have working here. THere is something wierd going on updating this one to github.

You could try manually editing the file

sudo visudo -f /etc/sudoers.d/RPI_Cam_Web_Interface 

It should just have the following in
# allow 'shutdown' command via web server
www-data ALL=/sbin/shutdown
www-data ALL=NOPASSWD: /sbin/shutdown
www-data ALL=/bin/date
www-data ALL=NOPASSWD: /bin/date

modsbyus, 2018-02-13:
I am unable to issue any sudo commands right now. They all give that error.

roberttidey, 2018-02-13:
I just tried doing an install and it went without error and is working OK

I did delete the RPi installer folder and cloned it fresh so it is possible that the update did something different with that file but it shouldn't make any difference.

modsbyus, 2018-02-13:
Any idea what could cause this. I wasn't having this problem before updating RPi Cam. I mounted the Pi SD card in an Ubuntu machine and the /etc/sudoers.d/RPI_Cam_Web_Interface looks right. There is another file in the sudoers.d directory 010_pi-nopassword. Is that also generated by your script?

roberttidey, 2018-02-13:
Script just copies the RPI_CAM version into sudoers.d 
The 010 file is the one that adds user pi into all sudo

At the moment I don't know what could have gone wrong.

I suggest you delete the RPI from sudoers.d from the Ubuntu machine. The Raspberry Pi should then boot OK with sudo active.

I'll have to try simulating an update process to see if I can break it.

modsbyus, 2018-02-13:
I did that and am able to sudo again. Then I removed the rpi cam folder in home and reinstalled. Everything is working great now. Thank you for the fantastic, quick support!