silvanmelchior/RPi_Cam_Web_Interface #397: Windows 7

CJiwan, 2018-02-15:
Hi,

  The web interface works perfectely on my RPi.
  I have one Windows 7 computer who can open without any problem the page : http://yourcamerip/min.php (of course with the IP of my Rpi).
  On a seconde Windows 7 computer I have all the time a script error 
   Line 197
  Char 7
  Error : Unable to set property 'disabled' of undefined or null reference

  Somebody can help me?

Christian

roberttidey, 2018-02-15:
Not sure where that is coming from.

Can't be in min.php as that only has a few lines.

Could be in script.js but then it's in the middle of an ajax_status and if there was an issue then it would be occurring with full web display as well.

Can I check are you seeing the error at the browser end or at the server (camera)?

You will get more info if you turn on developer tools in the browser and then access the min.php url

CJiwan, 2018-02-15:
Hi roberttidey,

  Finaly I find the solution.  On the 2 computer I have the same error but on the first one, IE are in "Silent mode" and not the second one.

  All works like I expect now.

  Thanks for your help

closed 2018-02-15 by CJiwan