silvanmelchior/RPi_Cam_Web_Interface #399: Security issue - datetime field does not sanitise input before passing to sudo

oleskiewicz, 2018-02-20:
Following #396, I have managed to execute a command on a Pi Zero W through time input field.

Time is changed through

```php
shell_exec("sudo date -s \"$timestr\"");
```

and `$timestr` is an unsanitised string from an AJAX request, passed directly to the `sudo` command.

Entering ``SOME DATE; `sudo shutdown now`;`` works and powers off the device, meaning other attacks can be performed.

This could be fixed by:

- only accepting input from a datetime-like widget
- sanitising the input on the backend

roberttidey, 2018-02-20:
good point. I'll look into sanitising the timestring before executing it.

roberttidey, 2018-02-20:
Latest version checks for valid date string and only executes if OK.

closed 2018-02-22 by oleskiewicz