silvanmelchior/RPi_Cam_Web_Interface #400: URL /cam_pic_new.php Not Found

smallzZz8, 2018-02-20:
So I was trying to get a URL to the direct feed of the camera for ISPY. I visited the wiki and was able to find 

> Now that an MJPEG stream has been implemented, you can configure iSpy with the following:
> 
> MJPEG URL tab within Video Source:
> 
> http://<IP>/cam_pic_new.php?

However that does not seem to work. When I visit that URL with my IP I am getting 

> Not Found
> The requested URL /cam_pic_new.php was not found on this server.

Does anyone have any suggestions on on how I can get the direct link to the feed for CCTV programs? Thanks!

roberttidey, 2018-02-20:
If you have installed into the default html subfolder then the URL would need to be
http://IP/html/cam_pic_new.php

Note that you can also include parameters on this to choose the frame rate
http://IP/html/cam_pic_new.php?pDelay=100000

where pDelay is time between frames in microseconds

smallzZz8, 2018-02-21:
Wow thank you thank you thank you!