silvanmelchior/RPi_Cam_Web_Interface #404: Images capture difference between "Record image"& "Record video start

fantyd, 2018-03-03:
The "Record image" button in the interface capture image in a wider image but the "Record video start" when capture motion or video in a smaller coverage area compare to "Record image".
Can i make the "Record video start" capture image same wide as "Record image" button?

roberttidey, 2018-03-03:
There are separate settings for still image and video resolutions under camera settings. So if you make those the same then the images will have the same coverage.

You cannot set the video width to be greater than 1920 as the GPU can't process those.

The raw camera sensor has a 4:3 aspect ratio. So if you want to capture the maximum field of view then you should choose the video / still to have the same ratio. If you choose a 16:9 ratio then it will crop some of the pixels off.

The live preview is always based on the video setting downscaled to have a width controlled by the preview width setting.

fantyd, 2018-03-03:
where is the setting for correcting to a maximum field of view?

closed 2018-03-05 by fantyd