silvanmelchior/RPi_Cam_Web_Interface #405: Extra explaination on Noise

fantyd, 2018-03-05:
The basic level of Noise setting is 1000, and if Noise level is added to 1009, in another words, the more high it is , the smoother changes and less trigger to false alarm. 

In the above, 
(1) Is there a Noise level limit for adding from 1000?
(2) If given a human and a animal(cat), how to set the Noise level amount for determine for not trigger false alarm when animal(cat) come in?

roberttidey, 2018-03-05:
There are 2 separate internal motion detect methods which are set by whether the noise level is below 1000 or >=1000. I always recommend using the second method (above 999) as it uses the vector data from the camera to determine which bits of the image are moving.

So when the Noise level is >= 1000 the second method is used and an internal filtering parameter is set = Noise - 999.

For the second method the Noise (-999) is used to provide a moving average of the measurement of motion. So Noise = 1010 means average the motion measurement over 11 frames. The higher the number then it takes longer to respond but will ignore rapid changes which are not sustained. There is no limit on this but values above say 1050 ( = 2 seconds) are unlikely to be useful.

A motion trigger occurs when the filtered changes are sustained for more than the start frame period setting. Longer values here help prevent false triggers but make it less responsive to rapid changes.

The Threshold is an important parameter. It sets the level of motion calculation in any one frame for the motion to be regarded as significant. Larger values mean it needs larger, faster moving objects to contribute to the trigger. So this would be one way of distinguishing between small pets and people.

The best values to use depend a lot on the overall scene, how far away objects are likely to be, lighting etc. It can be helpful in adjusting the values to display the basic motion detection values on the screen. You do this by including  %f and %c in the annotation template. This displays in real time the frame change count and the movement change measurement. By checking these values when particular motion is happening can help set the threshold and frame settings.

fantyd, 2018-03-06:
can I ask what is the mean for the word "counts vector" in the below sentence found in the guide?

The older internal method (<1000) counts vector changes in each frame which are above the 'noise setting' and if the total is above the threshold then it is a changed frame. If there is a sequence of changed frames more than the 'change frames to start' then a start trigger occurs.

roberttidey, 2018-03-06:
counts vector means the software calculates (counts) the amount of vector movement in each frame. The camera can provide vector data as well as pixel data because it uses the vector data as part of its video compression method. For each little square in the image it finds if this was similar to adjacent pixels in the previous frame and uses this to produce a vector describing how this little square is moving.

If you turn on the vector preview setting then you will get to see the vector data. It will be largely black but will show regions of data that are moving.