silvanmelchior/RPi_Cam_Web_Interface #407: Error in RaspiMJPEG

fantyd, 2018-03-20:
[image]
RPi Interface encounter error in RaspiMJPEG

closed 2018-03-20 by fantyd

kfordaccela, 2018-04-03:
I am also getting the same error on Raspbian Stretch and can't seem to find the issue, please let me know if there's some log info that I can help by sending.

roberttidey, 2018-04-03:
The original issue reported here was reported as closed by the poster.

The error dialog is caused by raspimjpeg not running. If you look in the scheduler log then it should give additional information,

I'd also encourage doing a clean install on a new raspbian stretch before installing any other software and just following the wiki steps.

kfordaccela, 2018-04-03:
I'm sorry to re-open this when it may not be an issue, this was tested on a fresh "image" of raspbian with only the "apt-get update" and "apt-get dist-upgrade" commands run. I will see about the scheduler log and raspimjpeg and getting that running.

roberttidey, 2018-04-03:
You can also check using the ps -A command to see what processes are running. There should be two entries for raspimjpeg

kfordaccela, 2018-04-03:
Issue came down to the fact that the camera wasn't correctly plugged in.

parskofamily, 2018-09-16:
Sorry to bring this up, but I have followed the same wiki instructions about 4 times, and this error keeps popping up.  Is there something wrong with it being outdated?  I don't have the experience to provide suggestions as to how to fix this.  When it was working, it was magical.  Pinewood Derby will never be the same!

roberttidey, 2018-09-18:
Check your camera connections at both ends. See if you can capture ok with raspivid

roberttidey, 2018-11-25:
Most of my cameras are on Pi Zeros which run fine.

Can you post bit of schedule log leading up to problem?

Also GPU memory should be 128MB, This should be the default but if it has got set to less then this can cause a problem.

parskofamily, 2018-11-28:
Yes, you are exactly right, and I'm wrong.  I have a dead camera.  I've swapped things out, loaded your software via the 8 or so commands on the wiki, and things are working. 

I'm not sure where things went wrong before, but I hadn't really made the proper effort to solve the problem.  I'm planning on stuffing these into legal Pinewood Derby cars.  I've finally been given a date for the race, which has "lit the fire" to get this problem solved.

Thanks for the ear and suggestions!  The software is awesome!!!

-Parsko